margelo/react-native-vision-camera #1597: How to crop a photo at a specific point

parice02, 2023-05-18:
### Question

Hi there!

I work with react-native-vision-camera which I like for its simplicity and quality of rendering. I would like to make a frame appear on the screen and when capturing the picture, only consider the part that is in the frame. I thought about **frameProcessor** but I'm stuck on the approach. Could someone please guide me?

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

samartsevigor, 2023-05-21:
Same problem

bglgwyng, 2023-05-22:
https://github.com/mrousavy/react-native-vision-camera/issues/1119#issuecomment-1173654329

I think you need to take a photo by `takePhoto` and manipulate the file created after then.
There's no way to crop a photo just when it's taken.

parice02, 2023-05-22:
Thank you for your different feedbacks. It's something like this that I wanted to do. 
 [image]

samartsevigor, 2023-05-22:
I noticed that if you take a photo, then its edges will be larger than what we see in the phone. This creates a large error

bryanltobing, 2023-06-22:
@parice02 do you able to resolve the issue?

bryanltobing, 2023-06-22:
@parice02 did you able to resolve the issue?

parice02, 2023-06-23:
> @parice02 did you able to resolve the issue?

Oh no!

bglgwyng, 2023-06-27:
https://github.com/bglgwyng/RNVC3Sandbox/blob/face-recognition-example/ios/FrameProcessors/FaceRecognizer/FaceRecognizerPlugin.swift#L17

Not sure if this code is helpful for you. It crops the faces by rectangles from the frame object.
If I understood it right and this is what you were looking for, I'll share the Android code also soon.

parice02, 2023-06-27:
Thanks for your reply. I'm thinking of writing some native code for that. I'm more interested in Android at the moment.

bglgwyng, 2023-06-28:
@parice02, I came across a situation where it required some time to create an example using my project code. Therefore, here I have decided to share only the cropping logic with you. https://gist.github.com/bglgwyng/4173f6096cf553f98909b0b07fd2e701
I believe you will be able to easily incorporate this code into your own project. It seems to be more related to an Android API matter rather than a specific issue with RNVC3. I suggest exploring alternative avenues where you can receive more helpful responses.

parice02, 2023-06-29:
Thanks for the help. I will customize it and integrate it into my code.

uzunhair, 2023-06-30:
> Я заметил, что если вы сделаете снимок, то его края будут больше, чем мы видим в телефоне. Это создает большую ошибку

Это из-за не правильных пропорций компонента камеры. Посмотрите какие фотографии у вас получаются после съемок, если к примеру их размер 1920 на 1080 (пропорции 1.77777) то на телефоне IphoneX с шириной 375 высота камеры должна быть (375*1.7777 ) ~666.6666. Если нарушена пропорция то по бокам или сверху будут появляться лишние элементы фото

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

JFYI; this should be handled in post-processing

closed 2023-09-30 by mrousavy