margelo/react-native-vision-camera #1599: Camera section goes black on some android devices

narada0923, 2023-05-19:
### Question

on some android devices camera goes black or nothing is shown. on ios everything works fine, but on some android device it goes black, and when I press capture button, photo is actually taken and was saved in photo library, but screen is still black. mostly these devices are android 11 or below. so is this library doesn't really support android 11 and below devices, or am I doing sth wrong?❓ 

### What I tried

const devices = useCameraDevices(); // i kept it empty to get best matching camera on devices
<Camera
    style={StyleSheet.absoluteFill}
    isActive
    device={devices.back}
    ref={ref}
    photo={true} 
/>

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

dawlatxai, 2023-05-22:
This issue happens to me also on Android 12. Did you find any solution?

cervebar, 2023-07-10:
Similar for us on Android 13 - black screen and simply not working, no error, camera permissions granted

erennyuksell, 2023-08-17:
same issue

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

mattlewer, 2023-08-31:
@mrousavy 

Still have this issue on a Pixel 4a with v3:
"react-native-vision-camera": "3.0.0-rc.8"
"react-native": "0.72.1"

Can confirm it works fine on iOS

closed 2023-09-01 by mrousavy