margelo/react-native-vision-camera #1600: ❓ Orientation switching on device rotation causes momentary preview freeze

ajp8164, 2023-05-19:
### Question

How can we prevent the preview video connection from freezing momentarily when switching orientation by changing the vision camera  orientation property. 

### What I tried

Only  looking at iOS at the moment..

I implemented the hook `useDeviceRotationSensor` as described in [624](https://github.com/mrousavy/react-native-vision-camera/issues/624). I also created an alternative local patch for iOS and providing  the same capability in swift. In both cases the result is the same. Rotate the device, the orientation for the video connection (AVCaptureVideoDataOutput) is changed. However, changing the orientation seems to  disconnect the current data writer, re-allocate buffers  and reconnect the data writer.

There must be a strategy to accomplish this since expo-camera does so as well as the iOS Camera app.

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

ajp8164, 2023-05-31:
In case others would like the camera to behave as in iOS Messages app (which is my goal)...

The momentary pause is caused by an orientation listener that is forcing a change to the camera output connection (buffer reallocation etc) when the device is rotated. To prevent this I have patched the iOS implementation to avoid the orientation listener under certain conditions and have added an output connection settings change to be done immediately prior to taking a photo and starting video recording. This works reasonably well - it's good enough for me to take to production. Android suffers a similar issue and so I'm working on that solution as well.  I'll issue a PR when I'm ready. In the mean time you can checkout my fork if you're interested - and make improvement suggestions ;-)

ajp8164, 2023-06-12:
Please see [PR 1619](https://github.com/mrousavy/react-native-vision-camera/pull/1619) for a native implementation.

bglgwyng, 2023-06-17:
I encountered the same error where the freezing issue arises only when the frame processor is connected. @ajp8164, could you kindly verify whether you have attached any frame processor to validate my observation? 
Additionally, the same behavior is found in [V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

bglgwyng, 2023-06-26:
Android works ok without freeze on orientation change. Tested on Galaxy S10.

bglgwyng, 2023-06-26:
I found that the problem was from the attached `videoOutput : AVCaptureVideoDataOutput`.
In both cases where I set `frameProcessor` or `video={true}`, then the freeze occurred.
Now I noticed that the frame processor internally processes video recording to do its job.
And it leads me to consider that the freeze is not a trivial bug 
because it's hard to say that the transition in the orientation of the videos being recorded has to be done smoothly.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy