margelo/react-native-vision-camera #1601: ❓ Camera: Native Module 'CameraView' was null! Did you run pod install?

pepperav, 2023-05-21:
### Question

Hi there 👋,
I'm struggling about this error. Both on Android and iOS. I've followed multiple guides, tutorials but still stuck on this error.

I'm using Expo and I'm trying to do the simple hello world about this library.

### What I tried

**package.json**
`
              "dependencies": {
                  "@types/react": "^18.2.6",
                  "@types/react-native": "^0.72.0",
                  "expo": "~48.0.15",
                  "expo-status-bar": "~1.4.4",
                  "react": "18.2.0",
                  "react-native": "0.71.8",
                  "react-native-reanimated": "^3.1.0",
                  "react-native-svg": "^13.9.0",
                  "react-native-vision-camera": "^2.15.4",
                  "typescript": "^5.0.4",
                  "vision-camera-dynamsoft-label-recognizer": "^0.3.3",
                  "expo-splash-screen": "~0.18.2"
                },
                "devDependencies": {
                  "@babel/core": "^7.20.0"
                },
`

**App.tsx**
`
      import React, { useEffect, useState } from 'react';
      import { SafeAreaView, StyleSheet, Text } from 'react-native';
      import { Camera, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera';
      import { ScanConfig, recognize } from 'vision-camera-dynamsoft-label-recognizer';
      import * as REA from 'react-native-reanimated';
      
      export default function App() {
	      const [hasPermission, setHasPermission] = useState(false);
	      const devices = useCameraDevices();
	      const device = devices.back;
      
	      useEffect(() => {
		      (async () => {
			      const status = await Camera.requestCameraPermission();
			      setHasPermission(status === 'authorized');
		      })();
	      }, []);
      
	      return (
		      <SafeAreaView style={styles.container}>
			      {device != null && hasPermission && (
				      <>
					      <Camera
						      style={StyleSheet.absoluteFill}
						      device={device}
						      isActive={true}
					      />
				      </>
			      )}
		      </SafeAreaView>
	      );
      }
      
      const styles = StyleSheet.create({
	      container: {
		      flex: 1,
	      },
	      barcodeText: {
		      fontSize: 20,
		      color: 'white',
		      fontWeight: 'bold',
	      },
      });
`

### VisionCamera Version

^2.15.4

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Jalson1982, 2023-05-23:
It seems it does not work with reanimated version 3

bryanltobing, 2023-06-13:
i'm currently getting this error as well

adam-a, 2023-07-07:
Getting this as well -- also not using reanimated

geekabhinavp, 2023-09-08:
guys any update on this?

geekabhinavp, 2023-09-11:
Found the fix, run `npx expo run:ios` to get it running on the emulator but you would get the devices as its an emulator, so connect the real device, and run the app using xcode itself it worked for me. As i think XCode does install pods as well when creating the build.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy