margelo/react-native-vision-camera #1602: GrRecordingContext.h file not found🐛

adityabishtedu, 2023-05-24:
### What were you trying to do?

i was building my ios application with react 0.71.7 with skia version : 0.1.191 and getting this header not found error

### What happened instead?

i recieved an import error
[image]

### Device

iPhone 14

### VisionCamera Version

3.0.0-rc-2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

AnomalousLLC, 2023-05-24:
Please reference the discussion thread: https://github.com/mrousavy/react-native-vision-camera/issues/1376

A patch was posted about a month ago.

AnomalousLLC, 2023-05-24:
Please reference the discussion thread: https://github.com/mrousavy/react-native-vision-camera/issues/1376

A patch was posted, over a month ago, that resolves this issue.

psyrenpark, 2023-05-25:
I am experiencing a build failure due to the mentioned reasons above. 
The current versions I am using are as follows:

```
"react-native": "0.72.0-rc.2",
"react": "18.2.0",
"react-native-reanimated": "^3.1.0",
"react-native-vision-camera": "^3.0.0-rc.2",
"@shopify/react-native-skia": "0.1.191",
"react-native-worklets": "https://github.com/chrfalch/react-native-worklets#d62d76c",
```

Please share your actual in-use package.json.
Thank you.

bglgwyng, 2023-05-26:
Please ensure that you set the version of '@shopify/react-native-skia' to 0.1.175 by pinning it.
Another option is to apply the patch, which can be found at https://github.com/mrousavy/react-native-vision-camera/issues/1376#issuecomment-1525731923.

tomgreco, 2023-06-27:
```
"@shopify/react-native-skia": "^0.1.175",
"react-native-worklets": "https://github.com/chrfalch/react-native-worklets#d62d76c",
"react-native-vision-camera": "^v3.0.0-rc.2",
"react-native": "0.70.10",
```

'include/gpu/GrRecordingContext.h' file not found

fukemy, 2023-07-28:
thanks @bglgwyng , worked to me
@tomgreco  did u tried to re-install pod

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy