margelo/react-native-vision-camera #1604: How do I properly unmount the camera with react-navigation❓

amjadbouhouch, 2023-05-27:
### Question

How do I properly unmount the camera with react-navigation using `useIsFocused()` ? 

### What I tried

I've tried the necessary unmounting methods from the docs using `isForeground` hook from the example and  `isFocussed` from react navigation. The is `isForeground` works but when I go back to previous screen `the green indicator still active`.

##

## Deps
```json
"dependencies": {
    "@react-navigation/native": "^6.1.6",
    "@react-navigation/native-stack": "^6.9.12",
    "react": "18.1.0",
    "react-native": "0.70.6",
    "react-native-safe-area-context": "^4.5.3",
    "react-native-screens": "^3.20.0",
    "react-native-vision-camera": "^2.15.4"
  },
```
## Screens
```tsx
export default function App() {
  return (
    <NavigationContainer>
      <Stack.Navigator>
        <Stack.Screen name="Home" component={HomeScreen} />
        <Stack.Screen name="Camera" component={CameraScreen} />
      </Stack.Navigator>
    </NavigationContainer>
  );
}

function HomeScreen() {
  const navigation = useNavigation();
  return (
    <View style={{flex: 1, alignItems: 'center', justifyContent: 'center'}}>
      <Button
        title="navigate to camera"
        onPress={() => navigation.navigate('Camera')}
      />
    </View>
  );
}
function CameraScreen({navigation}) {
  const devices = useCameraDevices();
  const device = devices.back;
  const isForeground = useIsForeground();

  const isFocussed = useIsFocused();

  if (device == null) {
    return null;
  }
  const isActive = isForeground && isFocussed;
  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={isActive}
    />
  );
}
```
## Device

Android

## Preview
https://github.com/mrousavy/react-native-vision-camera/assets/44619375/56729ec4-820e-4669-a5ec-7fb2fea930a2

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

sachitram-cv, 2023-06-07:
You can use navigation events (focus & blur).
https://reactnavigation.org/docs/navigation-events/

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

DeluxeOwl, 2023-10-10:
@amjadbouhouch can you please share the solution if you solved it?

Nice-zrd, 2023-12-20:
My project version is `"react-native": "0.72.6" , "react-native-vision-camera": "^3.6.17" ,`
I also encountered the same problem. Is there a solution available? After uninstalling the camera in Android, it still remains active. I have set the ` isActive={false} ` attribute。

JoyceQuerubino, 2024-06-17:
@amjadbouhouch can you please share the solution if you solved it?

amjadbouhouch, 2024-06-17:
@JoyceQuerubino I don't, I was testing and playing around.