margelo/react-native-vision-camera #1606: 🐛

tatianavazquez, 2023-05-30:
### What were you trying to do?

I'm implementing the flash functionality.

### Reproduceable Code

```tsx
CAMERA MODAL COMPONENT:

const CameraModal = ({visible, setVisible}) => {
  const [headerVisible, setHeaderVisible] = useState(false);

  const closeModal = () => {
    setVisible(false);
    setPhoto('');
  };

  const { t } = useTranslation();

  return (
    <Modal
      visible={visible}
      setVisible={() => setVisible(!visible)}
      transparent={false}
      modalStyle={{
        justifyContent: 'center',
        alignItems: 'center',
        backgroundColor: theme.colors.black,
        height: '100%',
      }}
    >
      <Spinner
        visible={showUntouchableOverlay}
        textContent={t('chat.loading')}
      />
      {photo ? (
        <PhotoUI
          image={{ uri: Platform.OS === 'ios' ? photo : `file://${photo}` }}
          nativeChatProps={nativeChatProps}
          onClose={closeModal}
        />
      ) : (
        <CameraUI />
      )}
    </Modal>
  );
};

export default CameraModal;

CAMERA UI:

const CameraUI = () => {
 const [photo, setPhoto] = useState('');
  const [cameraPermission, setCameraPermission] =
    useState<CameraPermissionRequestResult>();
  const camera = useRef<Camera>(null);
  const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>(
    'back',
  );
  const isPressingButton = useSharedValue(false);
  const devices = useCameraDevices();
  const cameraDevice = devices[cameraPosition];

  const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera);
  Reanimated.addWhitelistedNativeProps({
    zoom: true,
  });

  const setIsPressingButton = useCallback(
    (_isPressingButton: boolean) => {
      isPressingButton.value = _isPressingButton;
    },
    [isPressingButton],
  );

  const onMediaCaptured = useCallback((media: PhotoFile, type: 'photo') => {
    setPhoto(media.path);
  }, []);

  useEffect(() => {
    (async () => {
      const cameraPermissionStatus = await Camera.requestCameraPermission();
      setCameraPermission(cameraPermissionStatus);
    })();
  }, []);

  const [flash, setFlash] = useState<'off' | 'on'>('off');
  const supportsFlash = cameraDevice?.hasFlash ?? false;

  const onFlashPressed = useCallback(() => {
    setFlash(f => (f === 'off' ? 'on' : 'off'));
  }, []);

  return (
    <Container>
      <CameraHeader
        visible={headerVisible}
        flashOnPress={onFlashPressed}
        flash={supportsFlash ? flash : 'off'}
        arrowOnPress={() => setHeaderVisible(!headerVisible)}
        rawOnPress={() => console.log('TODO: Toggle raw')}
        liveOnPress={() => console.log('TODO: Toggle live')}
      />
        <ReanimatedCamera
          style={{
            width: screenWidth,
            height: screenHeight * 0.75,
          }}
          device={cameraDevice}
          isActive
          ref={camera}
          photo
          enableZoomGesture={false}
          animatedProps={cameraAnimatedProps}
          onError={error => {
            console.log({ error });
          }}
        />
      <Buttons>
        <ButtonsComponents.PhotoGalleryButton
          onPress={() => console.log('TODO: open gallery')}
        />
        <ButtonsComponents.CaptureButton
          camera={camera}
          onMediaCaptured={onMediaCaptured}
          cameraZoom={zoom}
          minZoom={minZoom}
          maxZoom={maxZoom}
          flash={supportsFlash ? flash : 'off'}
          enabled
          setIsPressingButton={setIsPressingButton}
        />
        <ButtonsComponents.ChangeCameraButton
          onPress={() => console.log('TODO: change camera')}
        />
      </Buttons>
    </Container>
  )
};

export default CameraUI;
```

### What happened instead?

Sometimes when I click on the flash icon, the camera turns black and nothing happens until I click on the icon again.
The console.log for 'error' on the Camera, gives me this error:

` LOG  {"error": [unknown/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSLocalizedFailureReason=An unknown error occurred (-12780), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x280d5bdb0 {Error Domain=NSOSStatusErrorDomain Code=-12780 "(null)"}}]}`

This error logs whenever I click on the camera icon.

### Device

iPhone 13

### VisionCamera Version

^2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

thunderousNinja, 2023-06-11:
I am facing the same issue, any updates? I am able to repro if I take two pictures right after each other.

tatianavazquez, 2023-06-11:
No, I don't have any updates and I don't know why this is happening... :/ where can we get help?

mrousavy, 2023-08-03:
Hey - hmm maybe the device does not support flash?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

DarshanSavaliyaGitHub, 2025-03-10:
Still facing below error.
```

[unknown/unknown: Error Domain=AVFoundationErrorDomain
Code=-11800 "The operation could not be completed" UserInfo={NSLocali
zedFailureReason=An unknown error occurred (-12780), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x300f15590 {Error Domain=NS
OSStatusErrorDomain Code=-12780 "(null)"}}]
```

I'm using this version.
 `"react-native-vision-camera": "4.6.3",`

mgulfam0722, 2025-04-24:
Still facing this issue:

```
[unknown/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSLocalizedFailureReason=An unknown error occurred (-12780), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x3024523d0 {Error Domain=NSOSStatusErrorDomain Code=-12780 "(null)"}}]
```

khanwilson, 2025-05-10:
Still facing this issue on iOS physical device both camera:
`[unknown/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSLocalizedFailureReason=An unknown error occurred (-12780), NSLocalizedDescription=The operation could not be completed, NSUnderlyingError=0x3024523d0 {Error Domain=NSOSStatusErrorDomain Code=-12780 "(null)"}}]`

"react-native-vision-camera": "^4.6.4"
"react-native": "0.75.4",

info.plist:

	<key>NSAppTransportSecurity</key>
	<dict>
		<key>NSAllowsArbitraryLoads</key>
		<true/>
		<key>NSAllowsLocalNetworking</key>
		<true/>
	</dict>

<key>NSCameraUsageDescription</key>
	<string>App requires access to your camera to capture photos and record videos for product sharing and virtual try-ons.</string>
	<key>NSMicrophoneUsageDescription</key>
	<string>App requires access to your microphone to record audio for video reviews and live shopping sessions.</string>

when call camera.current.startRecording still got that err.
Have any update :(

mgulfam0722, 2025-05-12:
Hello, 
The root cause of this issue, as per my observation, is that you might have unmounted the screen which contains the `<Camera />` component which causes this error. For example, using navigation methods like `navigation.replace()` or [`navigation.resetStack`](https://reactnavigation.org/docs/3.x/navigation-prop/#reset) will unmount the screen component and any preceding screen containing the `<Camera />` component will throw the mentioned error.

mgulfam0722, 2025-05-12:
> > Hello, The root cause of this issue, as per my observation, is that you might have unmounted the screen which contains the `<Camera />` component which causes this error. For example, using navigation methods like `navigation.replace()` or [`navigation.resetStack`](https://reactnavigation.org/docs/3.x/navigation-prop/#reset) will unmount the screen component and any preceding screen containing the `<Camera />` component will throw the mentioned error.
> 
> hmm, it not true with my situation. `<Camera />` is on a modal. it have call in a FormCreate Screen. And both of them still in stack. But still got err: Error Domain=AVFoundationErrorDomain Code=-11800. Still record without sound

Perhaps, you can verify if your screen component, containing the `<Camera />` component, is being unmounted using `useEffect` hook like this:
```
useEffect(() => {
    console.log('screen mounted!');
    return () => {
        console.log('Unmounted!');
    }
}, []);
```

ani-backend, 2025-07-25:
This error can also occur when you audio file is corrupted, empty, Doesn't exists or any unsupported configurations.
Example: Unsupported Sample rate, bit depth, To test try to replace with a test audio file..

williamgurzoni, 2025-09-02:
I'm facing the same problem with version `4.7.1`. I believe @mgulfam0722 is right on his observation. I also noticed the same situation. In my case this error happens when rotating the device (as I remount the component). Did you find a solution for this?

las-dungnm, 2025-09-25:
> I'm facing the same problem with version `4.7.1`. I believe [@mgulfam0722](https://github.com/mgulfam0722) is right on his observation. I also noticed the same situation. In my case this error happens when rotating the device (as I remount the component). Did you find a solution for this?

@williamgurzoni Have you found any solution yet? Let me know.