margelo/react-native-vision-camera #1609: Conflict with react-native-webrtc🐛

iamabdulhaseeb, 2023-05-30:
### What were you trying to do?

I am creating a video call app where users can record videos and participate in video calls. Initially, when I open the camera and record a video, it works fine. However, if I try to start a live stream afterwards (using the local stream track), it doesn't work as expected.

On the other hand, if I open the app and start a live stream first, it works successfully. However, when I turn off the live stream and then attempt to record a video, the react-native-vision-camera library doesn't function properly, and it displays a blank screen instead.

I am encountering this issue where the react-native-vision-camera library doesn't work as expected after using the live stream functionality. The camera view appears blank, preventing me from recording videos. I would appreciate any assistance in resolving this problem.

### What happened instead?

react native vision camera shows blank screen not camera.

### Device

Samsung A10s

### VisionCamera Version

^2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iamabdulhaseeb, 2023-05-30:
Okay so this happens when i try recording with back camera and webrtc with front camera,
when I do both with same camera either back or front then it works well

fukemy, 2023-05-31:
@iamabdulhaseeb could u  please share the way connect vision camera frame with local stream from react-native-webrtc?

quangnd1910, 2023-06-09:
I also have the same situation

mrousavy, 2023-08-03:
Hey! This is probably due to the libraries trying to take ownership of the Camera, when you set `isActive={true}` in VisionCamera it will try to open the Camera. If WebRTC then does the same, it will no longer have a Camera and you have to do `isActive={true}` again

mrousavy, 2023-08-03:
I'll try to handle interruptions better in V3, actually lemme leave this issue open

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy