margelo/react-native-vision-camera #1610: 🐛 latest version of gradle 1.6.20 is not supported

sabermed, 2023-06-03:
### What were you trying to do?

i have installed a latest version of android studio and nodejs v18 and setup everything correctly then i created a simple app with react-native(not Expo) that only shows button on the center the app running ok but when i installed react-native-vision-camera to use it after that whenever i run npx react-native run-android i get an error of unsupported ktolin gradle extention

### What happened instead?

whenever i type npx react-native run-android i get this error:

The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30

i think because of the react-native-vision-camera uses version 1.5.30 of kotlin but the version i have is 1.6.20

could you update the react-native-vision-camera to support the latest version of gradle?

### Relevant log output

```shell
The Android Gradle plugin supports only kotlin-android-extensions Gradle plugin version 1.6.20 and higher.
The following dependencies do not satisfy the required version:
project ':react-native-vision-camera' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30
```

### Device

Lenovo ideapad gaming 3 i5 RTX3050

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Jackson3195, 2023-06-05:
Can confirm same issue upgrading to RN 0.71.8

YaoGuangyuan, 2023-06-05:
Also have this issue.

a792883583, 2023-06-07:
> Can confirm same issue upgrading to RN 0.71.8
> 
> I did get around this by temporarily adding the following to the `android/build.gradle` file
> 
> [image]

Hello, I can call normally according to your Settings, but I found a new problem, when I started the debug mode, I found that the app would exit abnormally, and the app would still exit abnormally when I re-opened the app. Have you ever encountered this problem? Now my react-native is the latest version 0.71.8, kotlinVersion = "1.7.0". Thank you

eprice122, 2023-06-08:
I am seeing this issue as well

tahir-jamil, 2023-06-09:
i am having the same issue

SCJMENGMENG, 2023-06-15:
> Can confirm same issue upgrading to RN 0.71.8
> 
> I did get around this by temporarily adding the following to the `android/build.gradle` file
> 
> [image]

it`s useful for me

vaecebyZ, 2023-06-29:
It is very unfortunate that I am also experiencing the same issue，but it`s save my code and my time XD.

> Can confirm same issue upgrading to RN 0.71.8
> 
> I did get around this by temporarily adding the following to the `android/build.gradle` file
> 
> [image]

sahilrathi01, 2023-07-12:
i am also having the same issue
[image]

alanwong0922, 2023-07-13:
I have the same problem, I set the kotlinVerion as the latest version : 1.8.0, it still not work 

buildscript{ 
ext { 
`kotlinVerion:` 1.8.0
...
}
}

laurencetroyv, 2023-07-25:
Can confirm same issue on RN 0.72.3
[image]

Desintegrator, 2023-08-02:
> Can confirm same issue on RN 0.72.3 [image]

same

Mynameisjohndev, 2023-08-03:
I'm having the same problem, I've tried several things and so far nothing

marklanham, 2023-08-10:
> Can confirm same issue upgrading to RN 0.71.8
> 
> I did get around this by temporarily adding the following to the `android/build.gradle` file
> 
> [image]

This done the trick for now, thank you! 🙏

VitaminC1989, 2023-08-14:
> Hello, I can call normally according to your Settings, but I found a new problem, when I started the debug mode, I found that the app would exit abnormally, and the app would still exit abnormally [...]

I also encountered the same problem. I don't know how to solve it. I am currently using `reactotron-react-native` for development

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy