margelo/react-native-vision-camera #1611: ❓ get captureSession from CameraView

fukemy, 2023-06-07:
### Question

Hi, I want to transfer captureSession from Vision Camera to Webrtc camera, may be it's working. I am trying to do this

### What I tried

I tried to get `captureSession` by anyway, but seem not working.
Can u give me some advice @mrousavy 

### VisionCamera Version

v3 rc

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-06-07:
Hi, I don't understand the question. There's no interop between VisionCamera and WebRTC

closed 2023-06-07 by mrousavy

fukemy, 2023-06-08:
> Hi, I don't understand the question. There's no interop between VisionCamera and WebRTC

Hi, sorry for my bad English, let me explain more:
As the guide here, you may want other people to get each frame from Vision Camera in native way

 
[image]

 But I think I have other solution which is better, that is assign the video capture stream from Vision Camera `into` WebRTC stream, like this:
   `Webrtc` => `Stream` => `videoTrack` - `captureSession`
   Here I think I can assign the `captureSession` from `visionCamera` directly into `WebRTC` captureSession, so I would not to provide more code to handle the interop between VisionCamera and WebRTC
   
    I saw the swift class `CameraView` have `captureSession` property, all I want is just assign it into `Webrtc` stream
    
    In native IOS I can get `WebRTC` videotrack, is's contain `captureSession`, so I think if I do this, the `WebRTC` now can get the video directly from `VisionCamera`