margelo/react-native-vision-camera #1614: 🐛 Wrong orientation of Camera preview on iPadOS

saaymeen, 2023-06-08:
### What were you trying to do?

Render the preview of the front facing camera.

### Reproduceable Code

```tsx
import { StyleSheet } from 'react-native';
import { Camera, useCameraDevices } from 'react-native-vision-camera';

export { CameraScreen };

function CameraScreen(): JSX.Element {
	const devices = useCameraDevices();
	const device = devices.front;

	if (!device) return <></>;

	return (
		<Camera
			style={StyleSheet.absoluteFill}
			device={device}
			isActive={true}
			photo
		/>
	);
}
```

### What happened instead?

When the device is in landscape orientation, preview orientation is off by 90deg on landscape-right and off by -90deg on landscape-left.

### Device

iPad Pro (12,9", 3rd Generation)

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fede-moveshelf, 2023-06-12:
Same issue

babyjay-work, 2023-06-23:
I am having the same issue on an iPhone 7 (iOS 14.1) and an iPhone 12 (iOS 16.5) using react-native-vision-camera@2.15.4.  I did not have this issue on Expo SDK 45 and camera version 2.14.1.

When I have some free time, I may try to start at SDK 45 and upgrade one SDK at a time (and build new dev clients) to see when vision camera starts to break.

thebabyjay, 2023-06-23:
I am having the same issue on an iPhone 7 (iOS 14.1) and an iPhone 12 (iOS 16.5) using react-native-vision-camera@2.15.4. I did not have this issue on Expo SDK 45 and camera version 2.14.1.

When I have some free time, I may try to start at SDK 45 and upgrade one SDK at a time (and build new dev clients) to see when vision camera starts to break.

trashcoder, 2023-06-29:
same here with myiPhone 12

pdavid0, 2023-07-06:
Same issue here.
```
iPad mini 6th gen 
iOS 16.5.1

"react": "18.2.0",
"react-dom": "18.2.0",
"react-native": "^0.71.12",
"react-native-vision-camera": "^2.15.4",
"expo": "^48.0.19",
```

HemangiVekaria, 2023-09-04:
I am having same issue. Any solution?

eisodev, 2023-09-21:
Same issue here. iPhone 13 Pro.

On "react-native-vision-camera": "^2.15.6" because we can't upgrade our react native version yet.

I tried using:

```
import { useDeviceOrientation } from "@react-native-community/hooks";

const orientation = useDeviceOrientation();

orientation={orientation === 'portrait' ? 'portrait' : 'landscapeRight'}
```

But the orientation prop has no effect at all. I made sure the orientation prop updates correctly, which it does. I also tried forcing this prop to any of the values it can have, but still does nothing.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

JFYI; Orientation is quite compelx but if I have neough sponsors I'll work on it. Camera sensors are always 90deg

closed 2023-09-30 by mrousavy