margelo/react-native-vision-camera #1628: ❓ Not sure a bug, but I can not get the updated frame after drawing

fukemy, 2023-06-20:
### Question

### Description

I want to push frame from Vision Camera to WebRTC, before push frame, I did some draw on it

```

const paint = Skia.Paint()
paint.setColor(Skia.Color("red"))
const frameProcessor = useFrameProcessor((frame) => {
        'worklet';
    
        frame.drawRect({ x: 400, y: 400, width: 100, height: 100 }, paint)
        FrameProcessorPlugins.beautifulFace(frame)
    }, [paint])

```

Here is the captured video in my device:
[image]

But in implement of `beautifulFace`, I didnt see updated frame, here is the captured image:
[image]

I am debug using XCode, the code to debug is:
```
@objc(VisionFrameProcessorPlugin) => this called right afterFrameProcessorPlugins.beautifulFace(frame)
public class VisionFrameProcessorPlugin: NSObject, FrameProcessorPluginBase {
  @objc public static func callback(_ frame: Frame!, withArgs _: [Any]!) -> Any! {
    guard let imageBuffer = CMSampleBufferGetImageBuffer(frame.buffer) else {
      return nil
    }
    WebRTC.sendBuffer(frame.buffer)
...
```

Can u give me some advice

### What I tried

1. draw on to frame
2. check frame again

### VisionCamera Version

3.0.0 rc2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-07-20:
Hey - yep this is how it currently is in V3 - the Frame is only used for Preview. In the future this will also be used for Recording and TakePhoto, but I haven't got to this yet.

fukemy, 2023-07-21:
hell thanks you, I am tingyr to send rtc buffer to my rtcvideocapturerdelegate. Hope this feature avaiable soon 👍

closed 2023-07-21 by fukemy