margelo/react-native-vision-camera #1630: How to access Frame raw/buffer data❓

AmrElsersy, 2023-06-20:
### Question

I am using the FrameProcessor to receive frames continuously, but I don't want to run a native API, I want to access the raw data of the image/frame (the array of pixels of the image) through normal javascript code (I want to run a ONNX-react-native-js model through normal js not via native API)

I cannot find any attribute for buffer data in https://www.react-native-vision-camera.com/docs/api/interfaces/Frame#tostring
How can I do that ?

### What I tried

I tried to use FrameProcessor to catch frames

### VisionCamera Version

latest

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

fukemy, 2023-06-22:
did you have tried `frame.toByteArray()`?

Arnoldnuo, 2023-07-20:
> toByteArray

v2 doesn't have `toByteArray`, and v3 is still beta

mrousavy, 2023-08-03:
Will be possible in V3 :)

MateusHoffman, 2023-09-01:
@mrousavy how to convert a frame from useFrameProcessor callback to a base64?

mrousavy, 2023-09-04:
.toByteArray().

There's no byte64 of a raw GPU frame.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy