margelo/react-native-vision-camera #1635: Crash with `SIGSEGV` on Android

bglgwyng, 2023-06-29:
### What were you trying to do?

I wanted to run a camera with a frame processor in [V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

### Reproduceable Code

```tsx
I have not yet been able to solve the reproduction step. To initiate a collaborative effort in finding the solution, I am sharing this issue.

Nevertheless, I have come across a few small hints. Firstly, it seems that the frame processor may be relevant. The crash occurs only when I attach a frame processor to the camera. Additionally, the stack trace includes the presence of 'librnworklets.so'.

Furthermore, I believe that updating the `frameProcessor` prop could be a factor. When I did not update it, the crash did not occur. In V2, updating the frameProcessor prop, typically caused by modifying values in `deps` within `useFrameProcessor`, easily led to app crashes on both Android and iOS. Upon upgrading to V3, I discovered that the issue was resolved, particularly on iOS. However, there are still occasional crashes on Android, although not as frequently as in V2.
```

### What happened instead?

Run without a crash.

### Relevant log output

```shell
Crashed: Thread: SIGSEGV  0x0000000000000008
#00 pc 0x9a070 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#01 pc 0xa6300 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#02 pc 0xa6300 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#03 pc 0xa48ec libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#04 pc 0x87354 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#05 pc 0x74ce4 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#06 pc 0x5c7e8 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#07 pc 0x5c070 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#08 pc 0x1a19bc librnworklets.so (BuildId: 643768d061c79a4d4850ee762daa4c642174e953)
#09 pc 0x1a1954 librnworklets.so (BuildId: 643768d061c79a4d4850ee762daa4c642174e953)
#10 pc 0x1a18a0 librnworklets.so (BuildId: 643768d061c79a4d4850ee762daa4c642174e953)
#11 pc 0x1a1818 librnworklets.so (BuildId: 643768d061c79a4d4850ee762daa4c642174e953)
#12 pc 0x1a06c4 librnworklets.so (BuildId: 643768d061c79a4d4850ee762daa4c642174e953)
#13 pc 0x7c7dc libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#14 pc 0x87060 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#15 pc 0xa51a8 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#16 pc 0xa48ec libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#17 pc 0x87354 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#18 pc 0x74ce4 libhermes.so (BuildId: e5310494e026103da5c6017c6f68a9c1fa7262fa)
#19 pc 0x5c7e8 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#20 pc 0x5c070 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#21 pc 0x5bd34 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#22 pc 0x5e818 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#23 pc 0x5f1b8 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#24 pc 0x5f26c libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#25 pc 0x5f128 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#26 pc 0x5e7b8 libVisionCamera.so (BuildId: 9533f7f5b29e61ccec0b52b16d140796235a2072)
#27 pc 0x222244 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#28 pc 0x29c628 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#29 pc 0x218be8 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#30 pc 0x28520c libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#31 pc 0x764cdc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#32 pc 0x3e70f0 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#33 pc 0x3e1cec libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#34 pc 0x762dd4 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#35 pc 0x29cb50 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#36 pc 0x203994 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#37 pc 0x2000fc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#38 pc 0x2270ec libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#39 pc 0x763268 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#40 pc 0x775bf4 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#41 pc 0x203994 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#42 pc 0xa16ffc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#43 pc 0x2000fc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#44 pc 0xa16ffc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#45 pc 0x3d98e0 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#46 pc 0x2947b4 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#47 pc 0xa16ffc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#48 pc 0x3e1254 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#49 pc 0xa16ffc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#50 pc 0xa16ffc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#51 pc 0x3e1cc8 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#52 pc 0x202e94 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#53 pc 0x75dfe0 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#54 pc 0xa16ffc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#55 pc 0x203a14 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#56 pc 0x2000fc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#57 pc 0x2270ec libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#58 pc 0x763268 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#59 pc 0x775bf4 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#60 pc 0x203994 libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#61 pc 0xa16ffc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#62 pc 0x2000fc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
#63 pc 0xa16ffc libart.so (BuildId: 1724d2edb620afb8c3e8c5ea6c61ac19)
```

### Device

Galaxy S10

### VisionCamera Version

3.0.0-rc.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-08-03:
Can you share the entire logs from `adb logcat`?

bglgwyng, 2023-08-14:
@mrousavy I think it'll take some time to reproduce the bugs since made a huge migration and the old environment is lost. I'll test it once I fix the Android build with the latest RNVC V3!

mrousavy, 2023-08-14:
Cool yea I'm still in the middle of rewriting everything haha

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

bglgwyng, 2023-08-19:
@mrousavy  First, Thank you for your effort!

Brief report: I previously attempted to replicate this problem using a new project. My initial suspicion was that the crash resulted from updates to the 'frameProcessor' prop. Surprisingly, the small test app worked perfectly, suggesting that the issue might stem from domain-specific logic within my primary project.

The problem is that the Android build of my main project has been left broken for a period. Furthermore, the latest version of RNVC V3 was ok with a fresh RN project, hinting that the build failure may be linked to specific intricacies of my main project. Now I am trying to resolve the build problem from this point forward, but I cannot guarantee it to be done soon. 

I'll share the build problem in a new issue for further assistance.

bglgwyng, 2023-08-19:
I need to fix #1681 first. I've observed that the issue is not found in a new project + RNVC V3 installed.

mrousavy, 2023-08-21:
Exciting! V3 will close this then :) Thanks for testing!

closed 2023-09-01 by mrousavy

MajorTom007, 2023-11-14:
the same  issue , v 3.6.4
Sometimes the application crashes with this issue. On average, there is one crash per five scans.

Crashed: Thread: SIGSEGV  0x0000007cc639d7e1
#00 pc 0x128bac libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#01 pc 0x812f4 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#02 pc 0xa2600 libtflite_gmscore_jni.so (BuildId: abf371b4405232f9c7109f40a7426a23)
#03 pc 0xa17e4 libtflite_gmscore_jni.so (BuildId: abf371b4405232f9c7109f40a7426a23)
#04 pc 0x116474 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#05 pc 0x116484 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#06 pc 0x11601c libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#07 pc 0x112160 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#08 pc 0x1124a0 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#09 pc 0x82b98 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#10 pc 0x7aee4 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#11 pc 0x83360 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#12 pc 0x213a5 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#13 pc 0x7ac2c libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)
#14 pc 0x7ac38 libbarhopper_v3.so (BuildId: 73597d2de25bd3c29361db3c56dd9f18)

davidmrp, 2024-03-14:
Hi! The issue seems to be still present in latest `v3.9.1`. Reproduced on Redmi Note 11 Pro 5G (also this devices suffers "black screen" #2363, setting exposure=0 works). The app crashes after scanning a barcode and navigating to another screen and returning back to scanner screen and continue scanning barcodes. After 4-5 times doing this, the app crashes. Attached `device.json` with info of de camera device used and `adb_logcat.txt` with crash traces (crash occurs at `11:13:24.238`. Thanks!
 
[device.json](https://github.com/mrousavy/react-native-vision-camera/files/14600811/device.json)
[adb_logcat.txt](https://github.com/mrousavy/react-native-vision-camera/files/14600809/adb_logcat.txt)

[image]

Laulhus, 2024-03-18:
> Hi! The issue seems to be still present in latest `v3.9.1`. Reproduced on Redmi Note 11 Pro 5G (also this devices suffers "black screen" #2363, setting exposure=0 works). The app crashes after scanning a barcode and navigating to another screen and returning back to scanner screen and continue scanning barcodes. After 4-5 times doing this, the app crashes. Attached `device.json` with info of de camera device used and `adb_logcat.txt` with crash traces (crash occurs at `11:13:24.238`. Thanks!
> 
> [device.json](https://github.com/mrousavy/react-native-vision-camera/files/14600811/device.json) [adb_logcat.txt](https://github.com/mrousavy/react-native-vision-camera/files/14601027/adb_logcat.txt)
> 
> <img alt="crashlytics_trace" width="949" src="https://private-user-images.githubusercontent.com/70754615/312798854-e5e615eb-d1db-4773-8b13-5ff538821b83.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3NTQ1ODAsIm5iZiI6MTcxMDc1NDI4MCwicGF0aCI6Ii83MDc1NDYxNS8zMTI3OTg4NTQtZTVlNjE1ZWItZDFkYi00NzczLThiMTMtNWZmNTM4ODIxYjgzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDA5MzEyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3Y2MzZWE3YzE2MWMxM2VlZjEwNGM5N2ZmMzNlZTYzOGU3OTNmMTljNjQxOTgzM2U4YmMzYjg4MWMyMDRiNGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MPdSdGYDRN3aBeTd2z7zWaOQMQN6yY42t48I7UG4PaU">

Same here, I'm on 3.9.1 and I still experience this crash in different devices randomly after a few scans, I have the same stack trace

yousrasd, 2024-05-16:
Same here, experiencing this crash with v3.9.0