margelo/react-native-vision-camera #1636: 🐛 iOS archive fails due to VisionCamera/VisionCamera.modulemap' not found

tristanheilman, 2023-06-29:
### What were you trying to do?

I am trying to create a release build archive by utilizing Fastlane. Building and deploying using Xcode works perfectly fine. As soon as I attempt to create an archive to ship to TestFlight, I run into issues.

When building using Xcode, the DerivedData is coming from 

/myApp/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap

and I can see that this file exists.

When I run my fastlane script, the new location is under this file path

/myApp/Build/Products/Intermediates.noindex/

BUT VisionCamera is continuing to try and look in the Release-iphoneos/ folder.

### What happened instead?

The app does not build properly for Archives when using Fastlane. I could probably tell fastlane to not clean my DerivedData folders and things would work, but that is not what we want here. I expect VisionCamera to map the correct file paths.

### Relevant log output

```shell
[10:07:44]: $ set -o pipefail && xcodebuild -workspace Wridz.xcworkspace -scheme Wridz -destination 'generic/platform=iOS' -archivePath ./builds/release/Wridz\ 2023-06-29\ 10.07.44.xcarchive clean archive | tee /Users/treeeest/Library/Logs/gym/Wridz-Wridz.log | xcpretty
[10:07:44]: ▸ 2023-06-29 10:07:44.893 xcodebuild[23935:134193] DVTCoreDeviceEnabledState: DVTCoreDeviceEnabledState_Disabled set via user default (DVTEnableCoreDevice=disabled)
[10:07:47]: ▸ Clean Succeeded
[10:07:52]: ▸ Processing ResourceBundle-AccessibilityResources-React-Core-Info.plist
[10:07:52]: ▸ Copying /Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Intermediates.noindex/ArchiveIntermediates/Wridz/IntermediateBuildFilesPath/UninstalledProducts/iphoneos/AccessibilityResources.bundle/en.lproj
[10:07:52]: ▸ Touching AccessibilityResources.bundle (in target 'React-Core-AccessibilityResources' from project 'Pods')
[10:07:52]: ▸ Compiling pb_encode.c
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ Compiling pb_decode.c
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ Compiling pb_common.c
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ Compiling vlog_is_on.cc
[10:07:52]: ▸ Compiling utilities.cc
[10:07:52]: ▸ Compiling nanopb-dummy.m
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸ ❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[10:07:52]: ▸     Run script build phase 'Start Packager' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'Wridz' from project 'Wridz')
[10:07:52]: ▸     Run script build phase 'Bundle React Native code and images' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'Wridz' from project 'Wridz')
[10:07:52]: ▸     Run script build phase '[CP-User] [RNFB] Core Configuration' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'Wridz' from project 'Wridz')
[10:07:52]: ▸ ** ARCHIVE FAILED **
[10:07:52]: ▸ The following build commands failed:
[10:07:52]: ▸ 	CompileC /Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Intermediates.noindex/ArchiveIntermediates/Wridz/IntermediateBuildFilesPath/Pods.build/Release-iphoneos/nanopb.build/Objects-normal/arm64/pb_common.o /Users/treeeest/Projects/WridzRefactorV2/ios/Pods/nanopb/pb_common.c normal arm64 c com.apple.compilers.llvm.clang.1_0.compiler (in target 'nanopb' from project 'Pods')
[10:07:52]: ▸ (1 failure)
2023-06-29 10:07:44.893 xcodebuild[23935:134193] DVTCoreDeviceEnabledState: DVTCoreDeviceEnabledState_Disabled set via user default (DVTEnableCoreDevice=disabled)
▸ Clean Succeeded
▸ Processing ResourceBundle-AccessibilityResources-React-Core-Info.plist
▸ Copying /Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Intermediates.noindex/ArchiveIntermediates/Wridz/IntermediateBuildFilesPath/UninstalledProducts/iphoneos/AccessibilityResources.bundle/en.lproj
▸ Touching AccessibilityResources.bundle (in target 'React-Core-AccessibilityResources' from project 'Pods')
▸ Compiling pb_encode.c

❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found

▸ Compiling pb_decode.c

❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found

❌  fatal error: module map file '/Users/treeeest/Library/Developer/Xcode/DerivedData/Wridz-fxheznraswzysfeqzhticcgvvqwc/Build/Products/Release-iphoneos/VisionCamera/VisionCamera.modulemap' not found
[... 30 lines cut ...]
    Run script build phase 'Bundle React Native code and images' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'Wridz' from project 'Wridz')
[... cut ...]
### Device

iOS Archives

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nick-mylo, 2023-09-19:
@tristanheilman I am having the same issue. I cannot push a TestFlight build as a result. Do you have a workaround?

crpgoncalves, 2023-09-29:
@tristanheilman I have the same issue but using jenkins to archive the IPA. can you guys give some updates/tips abouts this topic ?
thanks

tristanheilman, 2023-09-29:
I'm still not 100% certain the reason this occurs but I have since managed to get around this. Also noting that this issue with _"modulemap not found"_ occurred with a few other packages my project utilizes. Some things you can try/check though...

- Close out Xcode app completely on the OS and try to utilize fast lane (or Jenkins) to build the app using command line tools; I noticed mixed results when Xcode app was closed
- Make sure your scheme set in the Xcode app matches the scheme you are using to build; AKA if fastlane is building for release, make sure your Xcode app is also set to the Release scheme.
- Close out of any other applications your Mac is using... I hit this problem exponentially more times when using my old MacBook Pro running an intel chip.. (constantly using multiple GB of swap memory). My new Mac Studio never touches swap.. and I have yet to see the issue occur on the new system...

Like I said, I'm no longer hitting this dumb issue.. If anything else comes to mind I will update or comment again. Sorry you guys are hitting the same issue but its gotta be something to do with cached files.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

coldSeattle, 2023-10-02:
same issue on mac os sonoma 14 and xcode 14 3 1

link82, 2023-10-06:
Same issue here with Sonoma / xcode 15. But I'm facing it only when trying to build and run on the simulator, works correctly when building for my device.

sahanK, 2024-01-18:
> Same issue here with Sonoma / xcode 15. But I'm facing it only when trying to build and run on the simulator, works correctly when building for my device.

I have the same issue. Were you able to find solutions for this ? I am using XCode 15.0.1 and building for iOS 17.0.1