margelo/react-native-vision-camera #1640: 🐛 Native Exception/Error does not get caught in useFrameProcessor worklet

jtklein, 2023-07-04:
### What were you trying to do?

I was trying to raise an Error/Exception in the native side of a frame processor plugin and catch it in the JS side of the useFrameProcessor worklet.

### Reproduceable Code

```tsx
I have created a repository that reproduces this error here:
https://github.com/jtklein/vision-camera-crash-reproduction
```

### What happened instead?

No crash. The native test error/exception that I introduced is raised and I can see it in the logcat/device logs of the Android and iOS device respectively. However, contrary to the [documentation](https://www.react-native-vision-camera.com/docs/guides/frame-processors-plugins-overview#exceptions) the error is never arriving in the catch block of the try/catch surrounding the frame processor function call in the useFrameProcessor worklet. So, even though the error is happening in the native code, I can not react to it in the JS side and show some kind of error state UI.

The logcat example below shows:
1) A log from the native Java cod of the Frame (so we know the native side is called correctly)
2) The test exception thrown in the native code.
3) A log from the JS code of the results of calling the frame processor function. (So we know the JS code stepped over the frame processor function line and did not catch)

What we do not see in this log: A log from the JS side catch block surrounding the function call.

### Relevant log output

```shell
Android logcat

2023-07-04 09:47:18.841 12896-6228  VisionCame...eproPlugin com.visioncameraerrorreproexample    D  2: 960 x 540 frame with format #35. Logging 0 parameters:0
2023-07-04 09:47:18.846 12896-6228  HermesVM                com.visioncameraerrorreproexample    I  JSI rethrowing JS exception: Exception in HostFunction: java.lang.RuntimeException: java.lang.Exception: This is a test exception
                                                                                                    
                                                                                                    Error: Exception in HostFunction: java.lang.RuntimeException: java.lang.Exception: This is a test exception
                                                                                                        at __xyz (native)
                                                                                                        at xyz (/Users/johannes/Documents/GitHub/vision-camera-crash-reproduction/src/index.tsx (8:7):1:34)
                                                                                                        at xyz (native)
                                                                                                        at _f (/Users/johannes/Documents/GitHub/vision-camera-crash-reproduction/example/src/App.tsx (23:43):1:92)
                                                                                                        at _f (native)
                                                                                                        at _f (/Users/johannes/Documents/GitHub/vision-camera-crash-reproduction/example/node_modules/react-native-vision-camera/src/hooks/useFrameProcessor.ts (28:21):1:425)
                                                                                                        at _f (native)
2023-07-04 09:47:18.853 12896-5982  ReactNativeJS           com.visioncameraerrorreproexample    I  'cvResults :>> ', undefined
```

### Device

Samsung G960F Android 10 / 29, iPhone 7 iOS 15.7.5

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-08-03:
Hey! Interesting, will try to fix that in V3 thanks for reporting g !

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy