margelo/react-native-vision-camera #1641: 🐛 Android build failure on v3

Romick2005, 2023-07-05:
### What were you trying to do?

[image]

### Reproduceable Code

```tsx
Any
```

### What happened instead?

App does not build on Android device.

### Relevant log output

```shell
* What went wrong:
A problem occurred configuring project ':app'.
> A problem occurred configuring project ':react-native-vision-camera'.
   > Could not determine the dependencies of null.
      > Could not resolve all task dependencies for configuration ':react-native-vision-camera:classpath'.
         > Could not find com.android.tools.build:gradle:.
           Required by:
               project :react-native-vision-camera
```

### Device

iPhone XS

### VisionCamera Version

RC3 v3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fukemy, 2023-07-06:
android still working in progress for rc3

mrousavy, 2023-07-07:
I said this three times already- android is still work in progress. The latest v3 is iOS only

closed 2023-07-07 by mrousavy

moloudayat, 2024-06-30:
I have the same problem with android. Is there any solution?