margelo/react-native-vision-camera #1642: 🐛 App does not build on iOS

Romick2005, 2023-07-05:
### What were you trying to do?

[image]

### Reproduceable Code

```tsx
Any
```

### What happened instead?

App does not build on iOS

### Relevant log output

```shell
'TensorFlowLiteObjC/TFLTensorFlowLite.h' file not found
```

### Device

iPhone XS

### VisionCamera Version

RC3 v3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fukemy, 2023-07-06:
Try adding the following to your podfile with modular headers.
```
pod 'TensorFlowLiteObjC', :modular_headers => true
pod 'TensorFlowLiteObjC/Metal', :modular_headers => true
pod 'TensorFlowLiteObjC/CoreML', :modular_headers => true
```

mrousavy, 2023-07-07:
You need to add the pods to the podfile, will be fixed in next rc

closed 2023-07-07 by mrousavy

Alexispap, 2023-07-12:
I added the pods, but the error remains.

ascholz81, 2023-07-13:
I have the same issue.

patras1, 2023-07-30:
I have the some issue, added the pods but still the some error

fukemy, 2023-07-31:
instead of claim the error, post the image or some error log

sudoMoshe, 2023-07-31:
to anyone who might be facing the same issue
I had the same problem with the m1 processor  expo SDK 49 RC3 v3
manually adding the TensorFlow lite framework to the project "Link Binary With Libraries" phase solved it 
[image]

willburstein, 2023-08-17:
I'm experiencing the same issue, but only once I incorporated Firebase with Firestore and use_frameworks