margelo/react-native-vision-camera #1645: Is it possible to hook into audio buffer for live transcription❓

adam-a, 2023-07-06:
### Question

I'm developing an application that aims to provide live transcription while the user records a video. The idea is to capture the user's audio, send it to a transcription service, and then overlay the transcribed text onto the recorded video. This application is particularly useful for situations like recording at Disney for a deaf friend.

My current challenge is finding a way to access the audio buffer so that I can send it to AWS Transcribe for transcription. I'm unsure if this approach is feasible or if there might be a better way to achieve my goal.

In summary, I want to create an app that can live transcribe user-recorded audio, overlay the transcription on the recorded video, and I need guidance on how to access the audio buffer for transcription using AWS Transcribe. Any advice or alternative suggestions would be greatly appreciated.

### VisionCamera Version

N/A

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-07-07:
Hey! Yea no currently you cannot access the Audio Buffer in JS. This shouldn't be too hard to implement though, the foundation is already there. If you want me to implement this for your business, contact me through my agency https://margelo.io

adam-a, 2023-07-07:
> https://margelo.io

Thanks! I can definitely attempt implementing it myself -- is it feasible to do with this library? I basically just need a starting point. You already answered my biggest question which is "is it possible" hahahha so thank you

fukemy, 2023-07-11:
yes it's possible

closed 2023-08-06 by adam-a