margelo/react-native-vision-camera #1646: 🐛 On android device unable to get reference of camera, on iOS it is working find.

zainlinux1989, 2023-07-07:
### What were you trying to do?

"react": "18.2.0",
"react-native": "0.71.10",
"react-native-vision-camera": "2.15.4",

```
const isFocused = useIsFocused();
<Camera
     ref={cameraRef}
     style={style.camera}
     device={device}
     isActive={isFocused}
     video={true}
     audio={true}
/>
```

### What happened instead?

for android case i am reloading view until it get the reference of camera.

### Device

Android

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

danstepanov, 2023-07-11:
Pls fix

fukemy, 2023-07-12:
Hi, which is `reference of camera.`, did you mean you are doing some code with `cameraRef` but first time it's undefined?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

zainlinux1989, 2023-09-01:
@mrousavy Thank you for doing your hard work, I have not checked but i hope this will be helpful for all. I will use it in our upcoming project