margelo/react-native-vision-camera #1655: 🐛 Unable to build using Mac M1

ronjunevaldoz, 2023-07-18:
### What were you trying to do?

CompileC /Users/ronjunevaldoz/Library/Developer/Xcode/DerivedData/Geo-dtbexgrcealbsbhhvwpzfqspydfm/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/VisionCamera.build/Objects-normal/x86_64/FrameProcessorRuntimeManager.o /Users/ronjunevaldoz/Projects/geoop.app/Geo/node_modules/react-native-vision-camera/ios/Frame\ Processor/FrameProcessorRuntimeManager.mm normal x86_64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
(1 failure)

### What happened instead?

error Command failed with exit code 1.

### Relevant log output

```shell
CompileC /Users/ronjunevaldoz/Library/Developer/Xcode/DerivedData/Geo-dtbexgrcealbsbhhvwpzfqspydfm/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/VisionCamera.build/Objects-normal/x86_64/FrameProcessorRuntimeManager.o /Users/ronjunevaldoz/Projects/geoop.app/Geo/node_modules/react-native-vision-camera/ios/Frame\ Processor/FrameProcessorRuntimeManager.mm normal x86_64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
(1 failure)
```

### Device

iPhone 14 Pro (Simulator)

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Vuuux, 2023-07-19:
Same here, found this, also followed all the solutions but didn't work:
https://github.com/mrousavy/react-native-vision-camera/issues/1450

ronjunevaldoz, 2023-07-19:
@Vuuux I've fixed it, by replacing the JSCRuntime to `<React-jsc/JSCRuntime.h>` then update the swift version to 5.2

Steps to fix
1. Update runtime to `<React-jsc/JSCRuntime.h>`
2. Update swift min version to 5.2  project.pbxproj `SWIFT_VERSION = 5.2;`
3. Create a patch so It will auto update the library before installation `yarn patch-package react-native-vision-camera`

closed 2023-07-19 by ronjunevaldoz