margelo/react-native-vision-camera #1657: ❓Abandoned version 2.x.x support while waiting for version 3

ronjunevaldoz, 2023-07-21:
### Question

I came across to this library, due to this is the recommended alternative to react-native-camera, which I trust the most. But after trying the latest release which `2.15.4` I found a lot of issues that are not being fixed or not being merged (community provided). Most of the users find it frustrating to use this recommended library but not being maintained properly. I'm always seeing reasons from the developer that before he fix it we better pay first or provide our own solution which is I think not a right manner when opening a community/open source project like this (I hate to see that this is recommended). 

### What I tried

manually patching up fixes `yarn patch-package`

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

latobibor, 2023-07-21:
The situation is this:
- previous camera packages are discontinued/abandoned
- the `react-native-camera-kit` has some traction recently, but I can't bet on it yet
- this package is maintained by an enthusiastic hobbyist

In this case we see the shortcomings of the open source experiment. He has every right to decline our requests, and we have no deterministic way to get what we want. I'm also frustrated with the situation, but it is not the author of the package I should be furious with but with Meta abandoning camera part of their `react-native` ecosystem.

You can't have hobbyists in the supply chain of a professional SDK. Probably you can get it through expo, but their `expo-camera` does not work in the `emulator` (because _f-you that's why_ if you ask me).

ronjunevaldoz, 2023-07-22:
@latobibor, it seems you're assuming I rely on this library. In reality, I'm just inquiring to make an informed decision about whether I should develop my own library. I generally prefer using existing solutions and not reinventing the wheel, but this particular scenario might require me to do so (ohh miracle).

mrousavy, 2023-08-03:
> I consistently encounter responses from the developer, suggesting that these issues will only be addressed if we pay for support or provide our own solutions. In my opinion, this approach is inappropriate for an open-source community project, and it saddens me to see it being recommended despite these challenges.

lol, huh? Do I owe you something my g? My time is valuable and I currently spend it on creating VisionCamera V3, a completely new version of this package which will fix pretty much 90% of the issues it currently has while also setting a new standard for cameras on mobile because of the Skia Frame Processors feature.

If you expect me to spend hours debugging issues **you** are experiencing and fixing them so **you** can release your product faster so **you** make money, then you're wrong, sorry.
The software is provided as is, and **it works great in my production apps and a lot of other apps out there** - sometimes they pay me to help them with some stuff.

Often the errors are because of some version incompatibilities, I'm addressing those in V3 as well.

closed 2023-08-03 by mrousavy