margelo/react-native-vision-camera #1658: error: package com.mrousavy.camera.frameprocessor does not exist 🐛

ranvirgorai, 2023-07-21:
### What were you trying to do?

Error : error: package com.mrousavy.camera.frameprocessor does not exist
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

[image]

### What happened instead?

Failing it gradle sync

### Device

Android

### VisionCamera Version

2.15.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2023-07-21 by ranvirgorai

erickcrus, 2025-01-10:
How do you solve this?

carlgrob5171989, 2025-04-25:
I am also facing the issue.