margelo/react-native-vision-camera #1662: ❓ Is there a way to run any version of the package with QR code reading in Nversions

latobibor, 2023-07-21:
### Question

Recently in the main discussion about `v3` someone claimed they had a working version of `v2` in `0.70.+` or `v3` pre-release with the QR code reader.

This grabbed my interest but I did not want to pollute

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

iBotPeaches, 2023-07-24:
Near all these questions you could find yourself. To lessen the load this maintainer takes - here you go:

 * No you cannot run Animated v3 on the current v2 version, an effort to allow that is here: https://github.com/mrousavy/react-native-vision-camera/pull/1623
 * You can use v2 and leverage the [plugin system](https://www.react-native-vision-camera.com/docs/guides/frame-processor-plugin-list#plugin-list) to use a frame processor to read QR codes right now.
  * We wrote our own [implementation](https://github.com/sourcetoad/vision-camera-plugin-barcode-scanner) that just wraps the Google MLKit so it can read QR, PDF417 and soon more
  * Yes its sucky that v2 has a good deal of crashes related to reanimated and the resolution from the perspective of the reanimated guys is "upgrade to v3". The perspective of this repo is "wait for v3 to be released". So the present time is a sucky state where you have a buggy v2 or an alpha quality v3.

latobibor, 2023-07-24:
Thanks for the answer!

> Near all these questions you could find yourself. To lessen the load this maintainer takes - here you go:
Partially the question was asked to help other people find these answer quicker. The other part was making sure I didn't miss someone being able to make this run.

Scattered information is one of the great sources of wasted engineering efforts.

metrix-hu, 2023-07-25:
Hi @latobibor!

The following four works just fine for us:

"react-native": "0.71.12",
"react-native-vision-camera": "2.15.4",
"react-native-reanimated": "2.17.0",
"vision-camera-code-scanner": "0.2.0"

However for vision-camera-code-scanner, we applied a patch on Android:

[vision-camera-code-scanner+0.2.0.patch](https://github.com/mrousavy/react-native-vision-camera/files/12155984/vision-camera-code-scanner%2B0.2.0.patch)

Ankit-Micron, 2023-07-26:
Hey

I am also trying to use vision-camera-code-scanner. The app works fine for iOS, but I get "**app:checkDebugDuplicateClasses**" error for Android.

Here are my versions: 
"react-native": "0.68.1",
"react-native-reanimated": "^2.8.0",
"react-native-vision-camera": "^2.14.1",
"vision-camera-code-scanner": "^0.2.0"

[image]

Any idea what can be the issue ?
Thanks

latobibor, 2023-07-27:
@metrix-hu Thanks for the answer and for struggling through the different versions (köszönjük szépen)! I will try out this!

xulihang, 2023-08-15:
Here is an example project used in the vision-camera-dynamsoft-barcode-reader which uses React Native 0.71: https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader/blob/main/example/package.json

rony-arnac, 2023-08-17:
Hi!
What about RN 72? 
@metrix-hu 's fix only supports up to RN 71

Thanks

metrix-hu, 2023-08-18:
@rony-arnac There is no new update yet for react-native-reanimated that supports RN 72. Its a bit of headache for me too, I am still on RN 71. There is no solution :(

latobibor, 2023-08-18:
This is the year when I encounter many important libraries being abandoned and with the open source supply chain looking increasingly amateurish. We can't expect people working on their free time to rapidly respond to big changes. Sadly folks at `react-native` do not seem to look behind them; if you are stuck with version `0.69` because a _ton_ of packages still do not support anything over `0.70+` then it is already no longer supported. In my opinion, they could just stop working altogether on newer versions of `react-native` and spend the engineering efforts helping to port important packages. But that's just my opinion.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.

latobibor, 2023-10-09:
Fantastic news! 👍

rrrgho, 2023-10-25:
it seems the onCodeScanned not return anything when I point the camera to a simple QR Code, I put console.log inside that callback function but it does nothing.
I followed all the docs on the MLKit and also RN Vision Camera QR Docs.

@mrousavy btw, thank you for your hard work, I Appreciate it

here is my Scaner page
const {hasPermission, requestPermission} = useCameraPermission();
  if (!hasPermission) {
    requestPermission();
  }

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      console.log('scanned: ', codes);
    },
  });

  useEffect(() => {
    console.log('rian');
  }, []);

  if (device == null) return <Text>No Camera</Text>;
  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={isActive}
      codeScanner={codeScanner}
    />
  );
  
  Here is my app/build.gradle
dependencies {
    // The version of react-native is set by the React Native Gradle Plugin
    implementation("com.facebook.react:react-android")
    implementation 'com.google.android.gms:play-services-mlkit-barcode-scanning:18.3.0'

Here is my src/main/AndroidManifest.xml

<manifest xmlns:android="http://schemas.android.com/apk/res/android">

    <uses-permission android:name="android.permission.CAMERA" />

    <uses-permission android:name="android.permission.INTERNET" />

    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:roundIcon="@mipmap/ic_launcher_round"
      android:allowBackup="false"
      android:theme="@style/AppTheme">      
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode"
        android:launchMode="singleTask"
        android:windowSoftInputMode="adjustResize"
        android:exported="true">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <meta-data
          android:name="com.google.mlkit.vision.DEPENDENCIES"
          android:value="barcode" />
    </application>
</manifest>