margelo/react-native-vision-camera #1664: 🐛 Can not build for android - Could not find fbjni-0.5.0-headers.jar

vahidvdn, 2023-07-24:
### What were you trying to do?

Without changing anything during the last few days, now out of a sudden, the build doesn't work. I get the some errors regarding maven.

### Reproduceable Code

```tsx
I stuck in the build time
```

### What happened instead?

Unable to build

### Relevant log output

```shell
* What went wrong:
Execution failed for task ':react-native-vision-camera:extractAARHeaders'.
> Could not resolve all files for configuration ':react-native-vision-camera:extractHeaders'.
   > Could not find fbjni-0.5.0-headers.jar (com.facebook.fbjni:fbjni:0.5.0).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/com/facebook/fbjni/fbjni/0.5.0/fbjni-0.5.0-headers.jar
```

### Device

Android 11

### VisionCamera Version

2.13.3 (also tested the latest version)

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

erksch, 2023-07-24:
I guess fbjni 0.5.0 is breaking (allowed by semantic version change)  but this library does not specify a correct version range.

```groovy
 if (REACT_NATIVE_VERSION < 71) {
      //noinspection GradleDynamicVersion
      extractHeaders("com.facebook.fbjni:fbjni:+:headers")
      //noinspection GradleDynamicVersion
      extractJNI("com.facebook.fbjni:fbjni:+")
```

https://github.com/mrousavy/react-native-vision-camera/blob/3ed019d018cce9ff20afb0fcc81146732419f17e/android/build.gradle#L291C1-L295C47

Gambitboy, 2023-07-24:
Are we able to temporarily tell react-native-vision-camera to download the previous version?

rfilipesantos, 2023-07-24:
If you change these two lines in **node_modules/react-native-vision-camera/android/build.gradle**

```
//noinspection GradleDynamicVersion
extractHeaders("com.facebook.fbjni:fbjni:0.4.0:headers")
//noinspection GradleDynamicVersion
extractJNI("com.facebook.fbjni:fbjni:0.4.0")
```
you can build your project successfully

erksch, 2023-07-24:
You can use [patch-package](https://www.npmjs.com/package/patch-package) to keep the changes around, but it will bloat your VSC a bit.

Gambitboy, 2023-07-24:
I only tried to alter the extractJNI, didn't think to also change extractHeaders.
Thanks. 👍🏻

pesohub, 2023-07-24:
> If you change these two lines in **node_modules/react-native-vision-camera/android/build.gradle**
>
> ```
> //noinspection GradleDynamicVersion
> extractHeaders("com.facebook.fbjni:fbjni:0.4.0:headers") [...]

It worked. As of now, version 0.5.0 of fbjni-java is not available.

alexandreerick, 2023-07-24:
> If you change these two lines in **node_modules/react-native-vision-camera/android/build.gradle**
>
> ```
> //noinspection GradleDynamicVersion
> extractHeaders("com.facebook.fbjni:fbjni:0.4.0:headers") [...]

It worked. Ty man.

antoine-wbr, 2023-07-24:
fbjni is releasing a 0.5.1 with header to fix this (https://github.com/facebookincubator/fbjni/issues/85#issuecomment-1647868528) but they have a worrisome comment about NDK compatibility (https://github.com/facebookincubator/fbjni/issues/85#issuecomment-1647826877). Shouldn't ```react-native-vision-camera``` refrain from specifying "any" version (+) ?

fendorio, 2023-07-24:
I have raised a PR to this repo which changes the import of fbjni to explicitly define version 4 as the dependency. This should really be the case anyway, to save this scenario from happening in the future.

#1666

sjcuello, 2023-07-24:
Same problem here !
Up to 21/07/2023 was working well !

mssoliveira, 2023-07-24:
> fbjni está lançando um 0.5.1 com cabeçalho para corrigir isso ( [facebookincubator/fbjni#85 (comentário)](https://github.com/facebookincubator/fbjni/issues/85#issuecomment-1647868528) ), mas eles têm um comentário preocupante sobre a compatibilidade do NDK ( [facebookincubator/fbjni#85 (comentário)](https://github.com/facebookincubator/fbjni/issues/85#issuecomment-1647826877) ). Não deveria `react-native-vision-camera`deixar de especificar "qualquer" versão (+)?

> 0.5.1 will be hitting Maven Central in the next minutes.
> 
> We'll follow up with more guidance on how to patch the affected libraries in the near future

here it's back to normal

fendorio, 2023-07-24:
It's great that a fix has been deployed upstream, however, this should library should still be patched to explicitly define the dependency version, as to avoid the same situation occurring again.

domthomhive, 2023-07-24:
Seems like the maven resource is missing.. 404 not found.  Anyone have any wider context on when/why this started happening? Can we expect this should be fixed on maven?

carolinaknoll, 2023-07-24:
@domthomhive I've also been reading through the issue in [fbjni's own thread](https://github.com/facebookincubator/fbjni/issues/85). From what I understand, when the owner released v0.5.0, he forgot to add the headers. So you can have a react native project that has the following lines, either on `android/app/build.gradle` or inside a dependency like `node_modules/package/app/build.gradle`.

 ```
 //noinspection GradleDynamicVersion
  extractHeaders("com.facebook.fbjni:fbjni:+:headers")
  //noinspection GradleDynamicVersion
  extractJNI("com.facebook.fbjni:fbjni:+")
```
  
These lines make it so that these projects can get the latest version of `fbjni` automatically, since they don't assign a particular version of `fbjni` to use. However, as `v0.5.0` doesn't have the headers, the project build fails because of it. 
  
The owner has released `v0.5.1` which has the headers. But as @fendorio said, this this same issue can happen in the future, because again, there are these lines of code saying "get the latest version automatically, regardless if it is available/has everything it needs or not".

Strangely, I thought that by using these lines we would start to get `v0.5.1` automatically so that the error would go away and the projects would build correctly again, which doesn't seem to be the case. My project is still failing to build. 🤔

sLurPPPeRsTaR, 2023-07-24:
hmm i tho i was the only one.. is there any update for this issue?

domthomhive, 2023-07-25:
@carolinaknoll thanks for the clarification. I noticed the 0.5.1 update had been published. If your builds still don't work it may be a caching issue. Worth busting the gradle caches... or run something like `./gradlew assembleDebug --refresh-dependencies` << worked for me.

tahaipek, 2023-07-25:
> If you change these two lines in **node_modules/react-native-vision-camera/android/build.gradle**
>
> ```
> //noinspection GradleDynamicVersion
> extractHeaders("com.facebook.fbjni:fbjni:0.4.0:headers") [...]

Thanks for the solution. It's working. I think version 0.5.1(fbjni) will fix it permanently.

fanr870606, 2023-07-25:
I added this section in the android/build.gradle file, It's a slightly cleaner approach.

```
project('react-native-vision-camera') {
    configurations.all {
        resolutionStrategy {
            force "com.facebook.fbjni:fbjni:0.4.0"
        }
    }
}
```

zzz08900, 2023-07-25:
> Strangely, I thought that by using these lines we would start to get `v0.5.1` automatically so that the error would go away and the projects would build correctly again, which doesn't seem to be the case. My project is still failing to build, as one of it's dependencies is still looking for `v0.5.0` and not `v0.5.1`. thinking
> 

Same here, I tried a clean-rebuild cycle a few hours ago but still got caught in the fbjni missing header problem.
But since 0.5.1 is published less than 24 hours ago I would think this is some CDN caching. Will check back in a few days - our build pipeline do clean-rebuild cycle every a few days to make sure all deps are up to date.

vahidvdn, 2023-07-26:
It's fixed for me too. For now, closing the issue. Thanks, everyone for your collaboration

closed 2023-07-26 by vahidvdn