margelo/react-native-vision-camera #1667: V3.0.0-rc.3 RN 0.72.2 android Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.🐛

lpfrenette, 2023-07-25:
### What were you trying to do?

Trying to upgrade from react-native-vision-camera v2 to v3

Installed version 3 as stated in this comment:
https://github.com/mrousavy/react-native-vision-camera/issues/1376#issuecomment-1618514195
```yarn add react-native-vision-camera@3.0.0-rc.3 react-native-worklets@https://github.com/chrfalch/react-native-worklets#3ac2fbb```

I also applied the patch using patch-package as stated in this comment :
https://github.com/mrousavy/react-native-vision-camera/issues/1376#issuecomment-1637021907

I tried every v3 branch with latest commit and i always have the same error.

### Reproduceable Code

```tsx
yarn react-native run-android
```

### What happened instead?

BUILD FAILED
```/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp/FrameHostObject.cpp:11:10: fatal error: 'WKTJsiHostObject.h' file not found```
```/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp/FrameProcessorRuntimeManager.h:12:10: fatal error: 'WKTJsiWorkletContext.h' file not found```

### Relevant log output

```shell
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/.cxx/Debug/6b6f422g/arm64-v8a'
  [1/11] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o
  /Users/lpfrenette/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/Users/lpfrenette/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVisionCamera_EXPORTS -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/../cpp -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactCommon -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactCommon/callinvoker -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -I/Users/lpfrenette/Github/mobile-app/node_modules/@shopify/react-native-skia/cpp/skia -isystem /Users/lpfrenette/.gradle/caches/transforms-3/03dccab32998b7f1f4d6aa70c3dc2129/transformed/jetified-react-android-0.72.1-debug/prefab/modules/jsi/include -isystem /Users/lpfrenette/.gradle/caches/transforms-3/03dccab32998b7f1f4d6aa70c3dc2129/transformed/jetified-react-android-0.72.1-debug/prefab/modules/reactnativejni/include -isystem /Users/lpfrenette/.gradle/caches/transforms-3/03dccab32998b7f1f4d6aa70c3dc2129/transformed/jetified-react-android-0.72.1-debug/prefab/modules/folly_runtime/include -isystem /Users/lpfrenette/.gradle/caches/transforms-3/34c2120d0d3cd8dc97826fc02c9e8fc4/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/include -isystem /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/example/node_modules/react-native-worklets/android/build/headers/rnworklets -isystem /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/example/node_modules/@shopify/react-native-skia/android/build/headers/rnskia -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -g  -fno-limit-debug-info -fPIC -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DFOLLY_HAVE_RECVMMSG=1 -DFOLLY_HAVE_PTHREAD=1 -DFOLLY_HAVE_XSI_STRERROR_R=1 -std=gnu++17 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o -c /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp/FrameHostObject.cpp
  /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp/FrameHostObject.cpp:11:10: fatal error: 'WKTJsiHostObject.h' file not found
  #include <WKTJsiHostObject.h>
           ^~~~~~~~~~~~~~~~~~~~
  1 error generated.
  [2/11] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  /Users/lpfrenette/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/Users/lpfrenette/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVisionCamera_EXPORTS -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/../cpp -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactCommon -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactCommon/callinvoker -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -I/Users/lpfrenette/Github/mobile-app/node_modules/@shopify/react-native-skia/cpp/skia -isystem /Users/lpfrenette/.gradle/caches/transforms-3/03dccab32998b7f1f4d6aa70c3dc2129/transformed/jetified-react-android-0.72.1-debug/prefab/modules/jsi/include -isystem /Users/lpfrenette/.gradle/caches/transforms-3/03dccab32998b7f1f4d6aa70c3dc2129/transformed/jetified-react-android-0.72.1-debug/prefab/modules/reactnativejni/include -isystem /Users/lpfrenette/.gradle/caches/transforms-3/03dccab32998b7f1f4d6aa70c3dc2129/transformed/jetified-react-android-0.72.1-debug/prefab/modules/folly_runtime/include -isystem /Users/lpfrenette/.gradle/caches/transforms-3/34c2120d0d3cd8dc97826fc02c9e8fc4/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/include -isystem /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/example/node_modules/react-native-worklets/android/build/headers/rnworklets -isystem /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/example/node_modules/@shopify/react-native-skia/android/build/headers/rnskia -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -g  -fno-limit-debug-info -fPIC -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DFOLLY_HAVE_RECVMMSG=1 -DFOLLY_HAVE_PTHREAD=1 -DFOLLY_HAVE_XSI_STRERROR_R=1 -std=gnu++17 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -c /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp/VisionCamera.cpp
  In file included from /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp/VisionCamera.cpp:3:
  /Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp/FrameProcessorRuntimeManager.h:12:10: fatal error: 'WKTJsiWorkletContext.h' file not found
  #include <WKTJsiWorkletContext.h>
           ^~~~~~~~~~~~~~~~~~~~~~~~
  1 error generated.
  [3/11] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o
  /Users/lpfrenette/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/Users/lpfrenette/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVisionCamera_EXPORTS -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/../cpp -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native-vision-camera/android/src/main/cpp -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactCommon -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactCommon/callinvoker -I/Users/lpfrenette/Github/mobile-app/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -I/Users/lpfrenette/Github/mobile-app/node_modules/@shopify/react-native-skia/cpp/skia -isystem /Users/lpfrenette/.gradle/caches/transforms-3/03dccab32998b7f1f4d6aa70c3dc2129/transformed/jetified-react-android-0.72.1-debug/prefab/modules/jsi/include -isystem /U [... cut ...]

Samsung Galaxy s8

### VisionCamera Version

3.0.0-rc.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

PopBot, 2023-07-25:
Thanks @lpfrenette, having a similar problem on iOS with 3.0.0-rc, where the `react-native-worklets` package cannot be found.

patras1, 2023-07-28:
Having the some issue...

mrousavy, 2023-07-31:
Latest V3 (#4) should work on Android now too

closed 2023-07-31 by mrousavy