margelo/react-native-vision-camera #1670: Cannot find interface declaration for 'VISION_EXPORT_SWIFT_FRAME_PROCESSOR'🐛

fukemy, 2023-07-27:
### What were you trying to do?

I I want to create customframeprocessor from my newProject, but I got this error

[image]

### Reproduceable Code

```tsx
I am sure to create bridging-header file:
```

### What happened instead?

I really dont know the problem

### Device

can not build

### VisionCamera Version

3.0.0 rc3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fukemy, 2023-07-28:
I use the rc2 then it's working, not sure why those method removed in rc3

fukemy, 2023-07-28:
now then i got this error, can u help @mrousavy 
[image]

```
yarn add react-native-vision-camera@3.0.0-rc.2
yarn add react-native-worklets@https://github.com/chrfalch/react-native-worklets#3ac2fbb
yarn add yarn add @shopify/react-native-skia@0.1.197

    "react": "18.2.0",
    "react-native": "0.72.3",
```

I am using

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

erickcrus, 2023-10-19:
any solution? i`m using VisionCamera V3 (3.4.1).

mrousavy, 2023-10-20:
You need to upgrade to 3.5.1

MartinTorresCh, 2024-05-28:
any solution? i'm using vision camera v4