margelo/react-native-vision-camera #1672: ❓ Take raw photo without additional system effects

AlekseyP18, 2023-07-29:
### Question

Hi

I am going to develop an application that will take a photo to give some advices to clients using AI to make analization of photo. To implement this logic correctly I'll have to make raw photo without any system effects that improve the photos.

As I understood it's possible using native language (like Swift)

Is it possible to make the raw photos without any system effects using this library?

I'll really appreciate for you answer and explanation!!!

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-08-03:
Hey do you want YUV, RGB or RAW (DNG) images?

AlekseyP18, 2023-08-18:
Hi, thanks for you answer. I was on a vacation, so I couldn't answer before, but still this question is really important for me.

Right now I am not sure about an exact format of an image that we will use. If it's possible to get an image without any system optimizations and effects I will get this project to implement it using react native, that I really like

I would ask you to explain how can I get this image in RGB or RAW format of images. Or if it's possible to get it directly in JPEG format (without system effects)

I am grateful for your time and contribution for react native community!!!

mrousavy, 2023-08-18:
Hey - yea I get what you're saying, but the only processing that's happening is in `takePhoto()` (RGB -> JPEG) and `recordVideo()` (YUV -> mp4/h264)

In a `useFrameProcessor` you get **raw, unprocessed** YUV frames in 4:2:0 8-bit format. Those are streamed in as the Camera is active.

Additionally, to take RAW (DNG) photos (`takePhoto()`), this should be possible soon with the new V3 API.

AlekseyP18, 2023-08-21:
So as I understood right now it's possible to take a photo without any system effects and optimizations just using - `takePhoto()` to get it in JPEG format.

Okey, thank you for you answer! I wish a good trip through developing this amazing library)

closed 2023-08-21 by mrousavy