margelo/react-native-vision-camera #1676: 'RNReanimated/REAIOSErrorHandler.h' file not found🐛

senolkeskin, 2023-08-02:
### What were you trying to do?

"react-native-reanimated": "^3.4.0",
"react-native-vision-camera": "^2.15.5",

When I run it through xcode I get the '**RNReanimated/REAIOSErrorHandler.h' file not found** error.

### Reproduceable Code

```tsx
#ifndef VISION_CAMERA_DISABLE_FRAME_PROCESSORS
  #if __has_include(<RNReanimated/NativeReanimatedModule.h>)
    #if __has_include(<RNReanimated/RuntimeManager.h>)
      #import <RNReanimated/RuntimeManager.h>
      #import <RNReanimated/RuntimeDecorator.h>
      #import <RNReanimated/REAIOSErrorHandler.h>
      #import "VisionCameraScheduler.h"
      #define ENABLE_FRAME_PROCESSORS
    #else
      #warning Your react-native-reanimated version is not compatible with VisionCamera, Frame Processors are disabled. Make sure you're using reanimated 2.2.0 or above!
    #endif
  #else
    #warning The NativeReanimatedModule.h header could not be found, Frame Processors are disabled. If you want to use Frame Processors, make sure you install react-native-reanimated!
  #endif
#endif

#import "FrameProcessorUtils.h"
#import "FrameProcessorCallback.h"
#import "../React Utils/MakeJSIRuntime.h"
#import "../React Utils/JSIUtils.h"
```

### What happened instead?

When I run it through xcode I get the '**RNReanimated/REAIOSErrorHandler.h' file not found** error.

### Relevant log output

```shell
'**RNReanimated/REAIOSErrorHandler.h' file not found**
```

### Device

Iphone 14

### VisionCamera Version

2.15.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Farzin-Firoozi, 2023-08-02:
Same problem here.

Farzin-Firoozi, 2023-08-02:
I fixed the issue by downgrading the react-native version to `0.71.7`.

toandn96, 2023-08-03:
Same problem here.

toandn96, 2023-08-03:
I fixed the issue by downgrading the react-native-reanimated version to 3.3.0.

caixingyue, 2023-08-03:
I also encountered the same problem, but it works fine on my real device. This problem occurs when running to the emulator, I have all the latest versions.

react-native-vision-camera/ios/Frame Processor/FrameProcessorRuntimeManager.mm:27:15 'RNReanimated/REAIOSErrorHandler.h' file not found

caixingyue, 2023-08-04:
> I also encountered the same problem, but it works fine on my real device. This problem occurs when running to the emulator, I have all the latest versions.
> 
> react-native-vision-camera/ios/Frame Processor/FrameProcessorRuntimeManager.mm:27:15 'RNReanimated/REAIOSErrorHandler.h' file not found

I delete node_modules, after re-installation. The whole exception is directly. I've tried installing a clean framework and still can't fix the issue. This seems to be related to the new version released a few days ago

nn1900, 2023-08-06:
as stated in https://www.react-native-vision-camera.com/docs/guides/frame-processors#disabling-frame-processors, I add `VISION_CAMERA_DISABLE_FRAME_PROCESSORS=1` to preprocessor definitions, however, it seems does not work and the error still shows up while rebuilding.

RaphBlanchet, 2023-08-06:
I seem to have worked my way around this issue by completely removing the code related to Frame Processors on iOS. Of course, in my case, I wasn't using them anyway, so removing the code temporarily is good enough for now.

To do this, add and configure [patch-packge](https://www.npmjs.com/package/patch-package) to your project and add [this patch](https://gist.github.com/RaphBlanchet/1e8ab14edb7ad49929487fec8360dd5f) in the `patches` folder at the root of your project. You will then have to re-run `yarn install && npx pod-install`

Hopefully we won't have to keep this workaround for long! 🤞🏼

chetanbhadarka, 2023-08-08:
> I fixed the issue by downgrading the react-native-reanimated version to `3.3.0`.

not working with RNAnimated version '3.3.0'

mrousavy, 2023-08-08:
Hi, this is a duplicate issue.

In the other issues I already mentioned that this will be fixed in V3. For now, I don't have the free time to fix that.

Shivampurbia, 2023-08-08:
> I used the @RaphBlanchet patch solution and it worked! (BTW I could not get Version 3.3.0 to solve the problem.)
> 
> [Ben Awad has an easy tutorial on how to use patches](https://www.google.com/search?q=npm+patch-package+example&oq=patch+npm&gs_lcrp=EgZjaHJvbWUqCAgFEAAYFhgeMgYIABBFGDkyCAgBEAAYFhgeMggIAhAAGBYYHjIICAMQABgWGB4yCAgEEAAYFhgeMggIBRAAGBYYHjIICAYQABgWGB4yBggHEEUYPNIBCTE1NTYzajBqOagCALACAA&sourceid=chrome&ie=UTF-8#fpstate=ive&vld=cid:6253f04c,vid:2AVs-Yh1bS8)

which version of RNanimated you have?

Shivampurbia, 2023-08-08:
> which version of RNanimated you have?

Thanks a lot man, it worked but i also had to comment this line,
` #import "../React Utils/MakeJSIRuntime.h"
`
otherwise it was giving me this error in multiple files whose call was initiated in this file.
`no template named 'optional' in namespace 'std'; did you mean 'folly::Optional'? `
this might be related to xCode version or something but i just commented that import alongside the changes that was mentioned in the patch.

novamaster-git, 2023-08-24:
If you guys are planning to use react-native-vision-camera with frame processor [i.e. QR scanner ]
then please use this version of packages:
```
react-native-reanimated: "2.17.0",
react-native-vision-camera: "2.15.4"
```
and it is working fine. 
for your information i am using 
React-Native: 0.70.12
ios: 16.6

khanakia, 2023-08-30:
Getting this error Unimplemented CameraView after i applied the patch https://gist.github.com/RaphBlanchet/1e8ab14edb7ad49929487fec8360dd5f

[image]

RaphBlanchet, 2023-08-30:
> Getting this error Unimplemented CameraView after i applied the patch https://gist.github.com/RaphBlanchet/1e8ab14edb7ad49929487fec8360dd5f
> 
> [image]

This is unrelated to the patch since it removes the code used by FrameProcessors, not CameraView. Make sure that you apply this patch to the correct version of react-native-vision-camera

khanakia, 2023-08-30:
I figured this out it's because of New Arch when i use this command it fails `RCT_NEW_ARCH_ENABLED=1 npx pod-install ios`

So i had to run this command instead of above command `npx pod-install ios`

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

PR #1759 (2023-09-18): feat: Reanimated v3 support in RNVisionCamera v2
## What

**This PR makes it possible to use RNVisionCamera v2 with Reanimated v3 (3.5.0+)** 🥳 

I've branched off from @mrousavy's PR (https://github.com/mrousavy/react-native-vision-camera/pull/1623) which bumps react-native version in the example app as well as does some general cleanup and integrated Reanimated v3 on top of it.

## Changes

* Move Camera thread runtime creation from C++ to JS using `Reanimated.createWorkletRuntime`
* Call worklets using `WorkletRuntime::runGuarded` which already supports error handling
* Consume Reanimated as [... cut]
closed 2023-09-30 by mrousavy