margelo/react-native-vision-camera #1677: ✨

MikhailMalinin-iOSDeveloper, 2023-08-02:
### What feature or enhancement are you suggesting?

Would be great to add support for scanning QRCodes from selected image in User's gallery

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Currently thinking to implement this feature from native code

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

closed 2023-08-03 by mrousavy

MikhailMalinin-iOSDeveloper, 2023-08-03:
Maybe we misunderstood each other,
I just want to convert image to 'Frame' model and call 'scanBarcodes' function

MikhailMalinin-iOSDeveloper, 2023-08-03:
cc @mrousavy

mrousavy, 2023-08-09:
Hey - uhm yea this is just not in the scope of VisionCamera at the moment as it requires a few more changes to the JSI setup.

I think it wouldn't be that tricky tbh, and I'm thinking about making Frame Processors a general library by extracting it from VisionCamera so that they can not only run on Camera streams, but on all kinds of media (off-screen photos/videos from gallery, `<Video>` components, `<Image>` component, etc.)