margelo/react-native-vision-camera #1678: 🐛 [iOS] Facing issue while building app

ravindraguptacapgemini, 2023-08-02:
### What were you trying to do?

When trying to build app, skip library is failing to build with below error log.

### Reproduceable Code

```tsx
Integrate latest RC version of react-native-vision-camera with RN 0.72.3 with new architecture enabled (fabric).

package.json dependencies:

"dependencies": {
    "@shopify/react-native-skia": "^0.1.197",
    "react": "18.2.0",
    "react-native": "0.72.3",
    "react-native-vision-camera": "^3.0.0-rc.4",
    "react-native-worklets": "github:chrfalch/react-native-worklets#3ac2fbb"
}

do pod install with command: bundle install && RCT_NEW_ARCH_ENABLED=1 bundle exec pod install
```

### What happened instead?

The iOS app is not building

### Relevant log output

```shell
▸ Compiling RCTModalHostViewComponentView.mm
▸ Compiling RCTInputAccessoryComponentView.mm
▸ Compiling RCTFabricComponentsPlugins.mm
▸ Compiling RCTComponentViewFactory.mm
▸ Compiling RCTActivityIndicatorViewComponentView.mm
▸ Compiling RCTKeyboardObserver.mm
▸ Compiling RCTI18nManager.mm
▸ Compiling RCTExceptionsManager.mm
▸ Compiling RCTDeviceInfo.mm
▸ Compiling RCTDevSplitBundleLoader.mm
▸ Compiling RCTDevSettings.mm
▸ Compiling RCTDevMenu.mm
▸ Compiling RCTDevLoadingView.mm
▸ Compiling RCTClipboard.mm
▸ Compiling RCTAppState.mm
▸ Compiling RCTActionSheetManager.mm
▸ Compiling RCTAccessibilityManager.mm
▸ Running script '[CP] Copy XCFrameworks'
▸ Building library libReact-RCTFabric.a
▸ Compiling RNSkMetalCanvasProvider.mm
▸ Compiling SkiaDomViewManager.mm
▸ Compiling SkiaDrawViewManager.mm
▸ Compiling RNSkManager.cpp

❌  /Users/ravindragupta/Desktop/Projects/AwesomeProject_Latest/ios/Pods/Headers/Public/RCT-Folly/folly/dynamic.h:63:10: 'boost/operators.hpp' file not found

#include <boost/operators.hpp>
                                                ^

▸ Compiling RNSkJsView.cpp
▸ Compiling RNSkDomView.cpp
▸ Compiling DrawingContext.cpp
▸ Compiling ConcatablePaint.cpp
▸ Building library libReact-CoreModules.a
    Run script build phase 'Start Packager' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'AwesomeProject_Latest' from project 'AwesomeProject_Latest')
    Run script build phase 'Bundle React Native code and images' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'AwesomeProject_Latest' from project 'AwesomeProject_Latest')

info 💡 Tip: Make sure that you have set up your development environment correctly, by running react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor 

node:internal/process/promises:279
            triggerUncaughtException(err, true /* fromPromise */);
            ^

[UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "Error: Failed to build iOS project. We ran "xcodebuild" command but it exited with error code 65. To debug build logs further, consider building your app with Xcode.app, by opening AwesomeProject_Latest.xcworkspace.".] {
  code: 'ERR_UNHANDLED_REJECTION'
}
```

### Device

iPhone 14

### VisionCamera Version

3.0.0-rc.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ravindraguptacapgemini, 2023-08-03:
@mrousavy Can you please check for this issue. Thanks

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy