margelo/react-native-vision-camera #1679: App crash iOS🐛

NimeshNB, 2023-08-04:
### What were you trying to do?

iOS app crashing if i'm using  1) **enableAutoStabilization:true** 2) **enableAutoDistortionCorrection:true**  
this two options below added code for **takephoto()** and Camera component.

```
   const photo = await this.cameraRef.current.takePhoto({
      qualityPrioritization:'quality',
      enableAutoStabilization:true,
      enableAutoDistortionCorrection:true
    });
```

and in Camera component :
```
    <Camera
              style={StyleSheet.absoluteFill}
              device={Device}
              ref={this.cameraRef}
              isActive={isactive}
              photo={true}
              enableHighQualityPhotos={true}
            />
```
   

### What happened instead?

The app crash with this error : AVFCapture: -[AVCapturePhotoOutput capturePhotoWithSettings:delegate:] + 752

### Device

iPhone 13 Pro

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

abdullahIsa, 2023-08-14:
@mrousavy i am not sure yet if this the cause on our side too but whenever user tries to take photo app will crash and show "Signal 6 was raised (null)" on iPhone.

abdullahIsa, 2023-08-16:
After pushing new update for testers in production iOS, removing enableAutoStabilization and enableAutoDistortionCorrection prevents the crash, i guess its same issue as this opened issue, any fix anyone ?

mrousavy, 2023-08-16:
Hmm maybe you need to check if the camera device in `useCameraDevices` actually supports stabilization ?

NimeshNB, 2023-08-17:
Yes removing **enableAutoStabilization** &  **enableAutoDistortionCorrection** prevents the crash
In ios quality is poor in android its fine.

abdullahIsa, 2023-08-27:
> Yes removing **enableAutoStabilization** & **enableAutoDistortionCorrection** prevents the crash In ios quality is poor in android its fine.

yep i agree quality is bad on iOS

NimeshNB, 2023-08-29:
> > Yes removing **enableAutoStabilization** & **enableAutoDistortionCorrection** prevents the crash In ios quality is poor in android its fine.
> 
> yep i agree quality is bad on iOS

Any solution to improve quality?

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy