margelo/react-native-vision-camera #1681: V3.0.0-rc.4 RN 0.72.3 android Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.🐛🐛

93asad, 2023-08-05:
### What were you trying to do?

Trying to upgrade from react-native-vision-camera v2 to v3

Installed version 3-RC4:
`yarn add react-native-vision-camera@3.0.0-rc.4 react-native-worklets@https://github.com/chrfalch/react-native-worklets#3ac2fbb @shopify/react-native-skia`

### Reproduceable Code

```tsx
yarn react-native run-android
```

### What happened instead?

[1/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o
FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o

 C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\memory:4434:5: error: static_assert failed due to requirement 'is_constructible<vision::FrameHostObject, facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>::value' "Can't construct object in make_shared"
      static_assert(is_constructible<_Tp, _Args...>::value, "Can't construct object in make_shared");

 D:/Developer/Projects/mCOPS/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.cpp:61:31: note: in instantiation of function template specialization 'std::make_shared<vision::FrameHostObject, facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrame, facebook::jni::JObject, void>::_javaobject *> &>' requested here
    auto frameHostObject = std::make_shared<FrameHostObject>(frame);

 C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\memory:2278:9: error: field of type 'vision::FrameHostObject' has private constructor
        : __value_(_VSTD::forward<_Args>(_VSTD::get<_Indexes>(__args))...) {}
          ^
  C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\memory:2362:9: note: in instantiation of function template specialization 'std::__compressed_pair_elem<vision::FrameHostObject, 1, false>::__compressed_pair_elem<facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &, 0>' requested here
          _Base2(__pc, _VSTD::move(__second_args),
          ^
  C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\memory:3635:16: note: in instantiation of function template specialization 'std::__compressed_pair<std::allocator<vision::FrameHostObject>, vision::FrameHostObject>::__compressed_pair<std::allocator<vision::FrameHostObject> &, facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>' requested here
              :  __data_(piecewise_construct, _VSTD::forward_as_tuple(__a),
                 ^
  C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\memory:4441:26: note: in instantiation of function template specialization 'std::__shared_ptr_emplace<vision::FrameHostObject, std::allocator<vision::FrameHostObject>>::__shared_ptr_emplace<facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>' requested here
      ::new(__hold2.get()) _CntrlBlk(__a2, _VSTD::forward<_Args>(__args)...);
                           ^
  D:/Developer/Projects/mCOPS/node_modules/react-native-vision-camera/android/src/main/cpp/JSIJNIConversion.cpp:120:28: note: in instantiation of function template specialization 'std::make_shared<vision::FrameHostObject, facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>' requested here
      auto hostObject = std::make_shared<FrameHostObject>(frame);
                             ^
  D:/Developer/Projects/mCOPS/node_modules/react-native-vision-camera/android/src/main/cpp/FrameHostObject.h:21:12: note: implicitly declared private here
    explicit FrameHostObject(const jni::alias_ref<JFrame::javaobject>& frame);

### Relevant log output

```shell
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `D:\Developer\Projects\mCOPS\node_modules\react-native-vision-camera\android\.cxx\Debug\4g2w03f2\arm64-v8a'
  [1/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o
  C:\Users\n1122168\AppData\Local\Android\Sdk\ndk\23.1.7779620\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe --target=aarch64-none-linux-android21 --sysroot=C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot -DVisionCamera_EXPORTS -ID:/Developer/Projects/mCOPS/node_modules/react-native-vision-camera/android/../cpp -ID:/Developer/Projects/mCOPS/node_modules/react-native-vision-camera/android/src/main/cpp -ID:/Developer/Projects/mCOPS/node_modules/react-native/ReactCommon -ID:/Developer/Projects/mCOPS/node_modules/react-native/ReactCommon/callinvoker -ID:/Developer/Projects/mCOPS/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -ID:/Developer/Projects/mCOPS/node_modules/@shopify/react-native-skia/cpp/skia -isystem C:/Users/n1122168/.gradle/caches/transforms-3/ef7b7082a4fed21bc2a02e3c84543346/transformed/jetified-react-android-0.72.3-debug/prefab/modules/jsi/include -isystem C:/Users/n1122168/.gradle/caches/transforms-3/ef7b7082a4fed21bc2a02e3c84543346/transformed/jetified-react-android-0.72.3-debug/prefab/modules/reactnativejni/include -isystem C:/Users/n1122168/.gradle/caches/transforms-3/ef7b7082a4fed21bc2a02e3c84543346/transformed/jetified-react-android-0.72.3-debug/prefab/modules/folly_runtime/include -isystem C:/Users/n1122168/.gradle/caches/transforms-3/68b86d627830190409a9d6f726c1e685/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/include -isystem D:/Developer/Projects/mCOPS/node_modules/react-native-worklets/android/build/headers/rnworklets -isystem D:/Developer/Projects/mCOPS/node_modules/@shopify/react-native-skia/android/build/headers/rnskia -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -g  -fno-limit-debug-info -fPIC -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DFOLLY_HAVE_RECVMMSG=1 -DFOLLY_HAVE_PTHREAD=1 -DFOLLY_HAVE_XSI_STRERROR_R=1 -std=gnu++17 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o -MF CMakeFiles\VisionCamera.dir\src\main\cpp\java-bindings\JFrameProcessor.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o -c D:/Developer/Projects/mCOPS/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.cpp
  In file included from D:/Developer/Projects/mCOPS/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.cpp:5:
  In file included from D:/Developer/Projects/mCOPS/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.h:7:
  In file included from C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\string:504:
  In file included from C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\string_view:175:
  In file included from C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\__string:57:
  In file included from C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\algorithm:643:
  C:/Users/n1122168/AppData/Local/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/include/c++/v1\memory:4434:5: error: static_assert failed due to requirement 'is_constructible<vision::FrameHostObject, facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JFrame, facebook::jni::JObject, void>::_javaobject *> &>::value' "Can't construct object in make_shared"
      static_assert(is_constructible<_Tp, _Args...>::value, "Can't construct object in make_shared");
      ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
Samsung Galaxy s20+

### VisionCamera Version

3.0.0-rc.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ascholz81, 2023-08-05:
I was able to obtain the same result.
React Native 0.72.3
Device - Emulator (Pixel 4) or Live Pixel XL 6
Build output: 
```

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/react-native-vision-camera/android/.cxx/Debug/5n51r232/arm64-v8a'
  [1/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [2/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrame.cpp.o
  [3/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JVisionCameraScheduler.cpp.o
  [4/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [5/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPluginHostObject.cpp.o
  [6/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o
  [7/13] Building CXX object CMakeFiles/VisionCamera.dir/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o
  [8/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  [9/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o
  [10/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o
  [11/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JVisionCameraProxy.cpp.o
  [12/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o
  [13/13] Linking CXX shared library ../../../../build/intermediates/cxx/Debug/5n51r232/obj/arm64-v8a/libVisionCamera.so
  FAILED: ../../../../build/intermediates/cxx/Debug/5n51r232/obj/arm64-v8a/libVisionCamera.so 
  : && /Users/adamscholz/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/Users/adamscholz/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -g  -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Qunused-arguments -Wl,--no-undefined -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/5n51r232/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JVisionCameraScheduler.cpp.o  /Users/adamscholz/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/lib/aarch64-linux-android/21/liblog.so  -landroid  /Users/adamscholz/.gradle/caches/transforms-3/1dc703403d0f8f29b4386f664e0a4308/transformed/jetified-react-android-0.72.3-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /Users/adamscholz/.gradle/caches/transforms-3/1dc703403d0f8f29b4386f664e0a4308/transformed/jetified-react-android-0.72.3-debug/prefab/modules/reactnativejni/libs/android.arm64-v8a/libreactnativejni.so  /Users/adamscholz/.gradle/caches/transforms-3/1dc703403d0f8f29b4386f664e0a4308/transformed/jetified-react-android-0.72.3-debug/prefab/modules/folly_runtime/libs/android.arm64-v8a/libfolly_runtime.so  /Users/adamscholz/.gradle/caches/transforms-3/1165844fcc849d9f3ad55af935b4624e/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so   -latomic -lm && :
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
  >>> referenced by VisionCameraProxy.cpp:35 (/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/react-native-vision-camera/android/src/main/cpp/VisionCameraProxy.cpp:35)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(vision::VisionCameraProxy::~VisionCameraProxy())
  >>> referenced by JFrameProcessor.cpp:39 (/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.cpp:39)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::JsiHostObject()
[... cut ...]
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.5/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 1m 54s
546 actionable tasks: 528 executed, 18 up-to-date
Configuration cache entry stored.

```

fangming4, 2023-08-07:
I have also same issue. When can this issue resolved? I need to use this module in a week.

bglgwyng, 2023-08-14:
In my working project, I encountered the same issue. 
After creating projects using both Expo and bare metal CLI, and installing only RNVC V3, the error was gone. 
I speculate that introducing additional build configurations to install other libraries may be causing this problem.

DeveloperMCD, 2023-08-15:
I'm having the exact same issue.

Zuxelus, 2023-08-17:
Had the same. Solved by deleting temp folders. I think it was \android\.gradle folder because I was not able to reproduce error any more.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

ascholz81, 2023-08-18:
AWESOME!  I'll get to it this weekend and report back.  Thanks to others as well, the note of "use_frameworks" is important and I do use firebase in this project so there's that.  :)

DeveloperMCD, 2023-08-18:
This error started happening for me again, even though I went back to 3.0.0-rc.4 and cleared out all my caches.

Here is what I'm getting:

> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a]
C/C++: ninja: Entering directory `/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/.cxx/Debug/j6t3e341/arm64-v8a'
C/C++: ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
C/C++: ld: error: undefined symbol: RNWorklet::JsiHostObject::JsiHostObject()
C/C++: ld: error: undefined symbol: RNWorklet::JsiHostObject::~JsiHostObject()
C/C++: ld: error: undefined symbol: RNWorklet::JsiWrapper::wrap(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWrapper*)
C/C++: ld: error: undefined symbol: RNWorklet::JsiHostObject::get(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&)
C/C++: ld: error: undefined symbol: RNWorklet::JsiHostObject::set(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&, facebook::jsi::Value const&)
C/C++: ld: error: undefined symbol: RNWorklet::JsiHostObject::getPropertyNames(facebook::jsi::Runtime&)
C/C++: ld: error: undefined symbol: typeinfo for RNWorklet::JsiHostObject
C/C++: ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnJsThread(std::__ndk1::function<void (facebook::jsi::Runtime&)>&&)
C/C++: ld: error: undefined symbol: RNWorklet::JsiWorkletContext::runtimeMappings
C/C++: ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__ndk1::function<void (RNWorklet::JsiWorkletContext*, facebook::jsi::Runtime&)>&&)
C/C++: ld: error: undefined symbol: RNWorklet::JsiWorkletContext::defaultInstance
C/C++: ld: error: undefined symbol: vtable for RNWorklet::JsiWorkletContext
C/C++: ld: error: undefined symbol: RNWorklet::JsiWorkletContext::~JsiWorkletContext()
C/C++: ld: error: undefined symbol: RNWorklet::JsiWorkletContext::JsiWorkletContext(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&, facebook::jsi::Runtime*, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>)

bglgwyng, 2023-08-19:
@DeveloperMCD Did you install Reanimated to dependencies?
Adding Reanimated to a clean project with only RNVC V3  installed caused this problem in my case.
However, I couldn't find the exact steps to reproduce it. Sometimes it occurred, sometimes it didn't.

ascholz81, 2023-08-19:
So I tried the new branch and it will not build in my project, but only for an unresolved reference in getOutputSizes.kt

[image]

DeveloperMCD, 2023-08-21:
The strange thing is, I was able for about a week to build my app just fine (in Android, not IOS) with 3.0.0-rc4. I couldn't get a cross platform barcode scanner working, but that's another story. I was able to successfully build my app, which has quite a few libraries. But ever since the day 3.0.0-rc5 came out, I started getting the errors listed in this thread. Very strange. Why is it having such errors with RNWorklets?

DeveloperMCD, 2023-08-22:
I think I found the problem! See this new bug report:

https://github.com/mrousavy/react-native-vision-camera/issues/1704

FouadMagdy01, 2023-08-22:
Firstly
From android studio
Go to sdk manager> android sdk > SDK Tools
And search for Cmake then install the 3.18.1 version 

Then go to android>app> build.gradle and go to this line

android {
.....
externalNativeBuild {
cmake {
version "3.18.1"

}
}
}

also make sure to install the latest version of react Native vision camera and react Native reanimated

DeveloperMCD, 2023-08-22:
I'm glad that helped you, @FouadMagdy01, but when I was having this problem, I tried both CMake 3.18 and 3.22 and it made no difference. I managed to compile the latest 3.0.0-rc5 (latest Vision Camera) and latest Reanimated by going in and updating React-Native-Worklets to the new repo where it was moved (not copied, MOVED) including a new repository, and a new name for the library! It is now called react-native-worklets-core.

Once I did that bit of work, I got a successful build! See my post about 2 posts up from this one. https://github.com/mrousavy/react-native-vision-camera/issues/1704

mrousavy, 2023-08-22:
react-native-worklets is now on npm as `react-native-worklets-core`! :)

closed 2023-08-22 by mrousavy