margelo/react-native-vision-camera #1682: V3.0.0-rc.4 RN 0.72.3 ios FrameHostObject.h not found🐛

ascholz81, 2023-08-05:
### What were you trying to do?

I want to use the frame processors 

### What happened instead?

iOS build fails while building VisionCamera with "FrameHostObject.h" not found
[image]

### Relevant log output

```shell
Showing All Errors Only

Build target VisionCamera of project Pods with configuration Debug

CompileC /Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/SkiaFrameProcessor.o /Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/react-native-vision-camera/ios/Skia\ Render\ Layer/SkiaFrameProcessor.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
    cd /Users/adamscholz/pdev/naxos/naxos-response-mobile/ios/Pods
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x objective-c++ -ivfsstatcache /Users/adamscholz/Library/Developer/Xcode/DerivedData/SDKStatCaches.noindex/iphoneos16.4-20E238-.sdkstatcache -target arm64-apple-ios11.4 -fmessage-length\=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit\=0 -std\=c++17 -stdlib\=libc++ -fobjc-arc -fmodules -gmodules -fmodules-cache-path\=/Users/adamscholz/Library/Developer/Xcode/DerivedData/ModuleCache.noindex -fmodules-prune-interval\=86400 -fmodules-prune-after\=345600 -fbuild-session-file\=/Users/adamscholz/Library/Developer/Xcode/DerivedData/ModuleCache.noindex/Session.modulevalidation -fmodules-validate-once-per-build-session -Wnon-modular-include-in-framework-module -Werror\=non-modular-include-in-framework-module -fmodule-name\=VisionCamera -Wno-trigraphs -fpascal-strings -O0 -fno-common -Wno-missing-field-initializers -Wno-missing-prototypes -Werror\=return-type -Wdocumentation -Wunreachable-code -Wno-implicit-atomic-properties -Werror\=deprecated-objc-isa-usage -Wno-objc-interface-ivars -Werror\=objc-root-class -Wno-arc-repeated-use-of-weak -Wimplicit-retain-self -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wduplicate-method-match -Wno-missing-braces -Wparentheses -Wswitch -Wunused-function -Wno-unused-label -Wno-unused-parameter -Wunused-variable -Wunused-value -Wempty-body -Wuninitialized -Wconditional-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wconstant-conversion -Wint-conversion -Wbool-conversion -Wenum-conversion -Wno-float-conversion -Wnon-literal-null-conversion -Wobjc-literal-conversion -Wshorten-64-to-32 -Wno-newline-eof -Wno-selector -Wno-strict-selector-match -Wundeclared-selector -Wdeprecated-implementations -Wno-c++11-extensions -Wno-implicit-fallthrough -DPOD_CONFIGURATION_DEBUG\=1 -DDEBUG\=1 -DCOCOAPODS\=1 -DPOD_CONFIGURATION_DEBUG\=1 -DDEBUG\=1 -DSK_METAL\=1 -DSK_GANESH\=1 -DVISION_CAMERA_ENABLE_FRAME_PROCESSORS\=true -DVISION_CAMERA_ENABLE_SKIA\=true -D_LIBCPP_ENABLE_CXX17_REMOVED_UNARY_BINARY_FUNCTION -DOBJC_OLD_DISPATCH_PROTOTYPES\=0 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS16.4.sdk -fstrict-aliasing -Wprotocol -Wdeprecated-declarations -Winvalid-offsetof -g -fvisibility-inlines-hidden -Wno-sign-conversion -Winfinite-recursion -Wmove -Wcomma -Wblock-capture-autoreleasing -Wstrict-prototypes -Wrange-loop-analysis -Wno-semicolon-before-method-body -Wunguarded-availability -index-store-path /Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Index.noindex/DataStore -iquote /Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/VisionCamera-generated-files.hmap -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/VisionCamera-own-target-headers.hmap -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/VisionCamera-all-non-framework-target-headers.hmap -ivfsoverlay /Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/all-product-headers.yaml -iquote /Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/VisionCamera-project-headers.hmap -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/VisionCamera/include -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/ReactCommon-Samples/ReactCommon_Samples.framework/Headers -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/ReactCommon/ReactCommon.framework/Headers/react/nativemodule/core -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/ReactCommon-Samples/ReactCommon_Samples.framework/Headers/platform/ios -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/React-NativeModulesApple/React_NativeModulesApple.framework/Headers -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/React-graphics/React_graphics.framework/Headers/react/renderer/graphics/platform/ios -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/ios/Pods/Headers/Public -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/ios/Pods/Headers/Public/FBLazyVector -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/ios/Pods/Headers/Public/RCTRequired -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/ios/Pods/Headers/Public/React-callinvoker -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/ios/Pods/Headers/Public/React-runtimeexecutor -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/ios/Pods/Headers/Public/hermes-engine -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/XCFrameworkIntermediates/react-native-skia/Headers -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/XCFrameworkIntermediates/react-native-skia/Headers -I/Users/adamscholz/Library/Developer/Xcode/DerivedData/naxos-assikdtcamslhmeoezmabigzimzk/Build/Products/Debug-iphoneos/XCFrameworkIntermediates/react-native-skia/Headers -I../../node_modules/react-native-vision-camera/cpp -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/modules -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/src -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/android -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/codec -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/config -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/core -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/docs -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/effects -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/encode -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/gpu -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/pathops -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/ports -I/Users/adamscholz/pdev/naxos/naxos-response-mobile/node_modules/@shopify/react-native-skia/cpp/skia/include/private -I/Users/ada [... cut ...]
simulator or live iPhone XR

### VisionCamera Version

v3.0.0-rc4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fangming4, 2023-08-07:
I have issue like this
Failed to install Frame Processor JSI bindings!

how can I fix it?

ascholz81, 2023-08-19:
I was able to get around this issue and the next "not found" by adding the following to my podfile
```
    # needed for react-native-vision-camera https://github.com/mrousavy/react-native-vision-camera/issues/1413#issuecomment-1604205944
    installer.pods_project.targets.each do |target|
      target.build_configurations.each do |config|
        config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native/ReactCommon" '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native-vision-camera/ios" '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native-vision-camera/ios/Frame Processor" '
        config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native-skia/cpp/api/third_party" '
      end
    end
```
However, I still have a build error in another file that I can't get around.  
[image]

pyc92, 2023-08-23:
> I have issue like this Failed to install Frame Processor JSI bindings!
> 
> how can I fix it?

same here

dahlmo, 2023-08-30:
The "Failed to install Frame Processor JSI bindings!" error seems to be resolved in [rc.8 (link)](camera/releases/tag/v3.0.0-rc.8). Recommend to always check if there's a newer RC out before reporting @pyc92 .

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy