margelo/react-native-vision-camera #1687: [Android] Flash light turn off when its used with frameprocessor🐛

ravirajcm, 2023-08-10:
### What were you trying to do?

I am trying to use <Camera> with frameprocessor also setting torch as 'on' with "useScanBarcodes" as QR to read QR codes.
But as soon as it detect QR it turn of the flash light even I have torch set as 'on'.

### Reproduceable Code

```tsx
<Camera
        ref={CameraRef}
        torch={flash}
        device={device}
        frameProcessor={frameProcessor}
        frameProcessorFps={5}
        isActive={!shouldPauseCamera}
      />
```

### What happened instead?

It should keep torch on when torch='on' and off when torch='off'

### Device

Samsung S22 plus

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

mrousavy, 2023-09-30:
Hey - I'm tracking Flash in this feature request/issue now: #1890  ✨ 

Make sure to upvote or sponsor to support this feature, and leave a comment if you have any additional thoughts/ideas.