margelo/react-native-vision-camera #1692: Task ':@react-native-camera-roll_camera-roll:compileDebugJavaWithJavac' uses this output of task ':react-native-camera-roll_camera-roll:compileDebugAidl' without declaring an explicit or implicit dependency.

DeveloperMCD, 2023-08-14:
### What were you trying to do?

Trying to compile my app -- which used to compile just fine. Problems started when I tried to use FrameProcessors. 
Specifically, these libraries: react-native-reanimated, vision-camera-code-scanner

I haven't been able to build my app successfully since I first started implementing "Hello World" with that vision-camera-code-scanner.

### Reproduceable Code

```tsx
npx react-native start
```

### What happened instead?

Gradle detected a problem with the following location: '/home/matthew/dev/rn/stackbayapp/node_modules/@react-native-camera-roll/camera-roll/android/build/generated/aidl_source_output_dir/debug/out'. Reason: Task ':@react-native-camera-roll_camera-roll:compileDebugJavaWithJavac' uses this output of task ':react-native-camera-roll_camera-roll:compileDebugAidl' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed. Please refer to https://docs.gradle.org/7.5.1/userguide/validation_problems.html#implicit_dependency for more details about this problem.

### Relevant log output

```shell
https://scans.gradle.com/s/zmfcpg75hqhem
```

### Device

N/A

### VisionCamera Version

2.15.6

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Dhiraj-j, 2023-08-16:
> ### What were you trying to do?
> Trying to compile my app -- which used to compile just fine. Problems started when I tried to use FrameProcessors. Specifically, these libraries: react-native-reanimated, vision-camera-code-scanner
> 
> I haven't been able to build my app successfully since I first started implementing "Hello World" with that vision-camera-code-scanner.
> 
> APOLOGIES FOR THE TITLE, which makes it seem like an issue with react-native-camera-roll. Here is where my compile is crashing: **What went wrong: Execution failed for task ':react-native-vision-camera:configureCMakeDebug[arm64-v8a]'.**
> 
> I've been receiving many errors, sometimes focused on "react-native-camera-roll" (see below error output) or other libraries touching on the camera. The thing is, I haven't touched that library (camera-roll) so it can't be something in that specific library causing the actual problem. It might be related to "reanimated", but this project is not "off the hook". Here is what I mean:
> 
> Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081 Picked up JAVA_TOOL_OPTIONS: -Xms2048m -Xmx6096M Note: /home/matthew/dev/rn/stackbayapp/node_modules/@react-native-camera-roll/camera-roll/android/src/main/java/com/reactnativecommunity/cameraroll/CameraRollModule.java uses or overrides a deprecated API. Note: Recompile with -Xlint:deprecation for details. Note: /home/matthew/dev/rn/stackbayapp/node_modules/@react-native-camera-roll/camera-roll/android/src/main/java/com/reactnativecommunity/cameraroll/CameraRollPackage.java uses unchecked or unsafe operations. Note: Recompile with -Xlint:unchecked for details. Note: /home/matthew/dev/rn/stackbayapp/android/app/src/main/java/com/stackbay/NetworkChangeReceiver.java uses or overrides a deprecated API. Note: Recompile with -Xlint:deprecation for details.
> 
> FAILURE: Build completed with 2 failures.
> 
> ## 1: Task failed with an exception.
> * What went wrong:
>   Execution failed for task ':react-native-vision-camera:configureCMakeDebug[arm64-v8a]'.
> 
> > Picked up JAVA_TOOL_OPTIONS: -Xms2048m -Xmx6096M
> 
> * Try:
> 
> > Run with --stacktrace option to get the stack trace.
> > Run with --info or --debug option to get more log output.
> > Run with --scan to get full insights.
> > ==============================================================================
> 
> ## 2: Task failed with an exception.
> * What went wrong:
>   java.lang.StackOverflowError (no error message)
> 
> And there's this:
> 
> "#1260 I ran into the same issue in new project with latest RN 0.72 + react-native-vision-camera react-native-vision-camera v2 requires react-native-reanimated v2.x.x which does not support new RN 0.72 support for 0.72 is only available since RNR v3.3.0 ([software-mansion/react-native-reanimated#4668](https://github.com/software-mansion/react-native-reanimated/issues/4668) (comment)), but you can't use RNR3 with react-native-vision-camera (#1210 (comment))"
> 
> ### Reproduceable Code
> ```tsx
> npx react-native start
> ```
> 
> ### What happened instead?
> Gradle detected a problem with the following location: '/home/matthew/dev/rn/stackbayapp/node_modules/@react-native-camera-roll/camera-roll/android/build/generated/aidl_source_output_dir/debug/out'. Reason: Task ':@react-native-camera-roll_camera-roll:compileDebugJavaWithJavac' uses this output of task ':react-native-camera-roll_camera-roll:compileDebugAidl' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed. Please refer to https://docs.gradle.org/7.5.1/userguide/validation_problems.html#implicit_dependency for more details about this problem.
> 
> ### Relevant log output
> ```shell
> https://scans.gradle.com/s/zmfcpg75hqhem
> ```
> 
> ### Device
> N/A
> 
> ### VisionCamera Version
> 2.15.6
> 
> ### Additional information
> * [ ]  I am using Expo
> * [x]  I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
> * [x]  I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
> * [x]  I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Have you find any solution?

DeveloperMCD, 2023-08-29:
This
"https://github.com/mrousavy/react-native-vision-camera/issues/1260 I ran into the same issue in new project with latest RN 0.72 + react-native-vision-camera react-native-vision-camera v2 requires react-native-reanimated v2.x.x which does not support new RN 0.72 support for 0.72 is only available since RNR v3.3.0 (https://github.com/software-mansion/react-native-reanimated/issues/4668 (comment)), but you can't use RNR3 with react-native-vision-camera (https://github.com/mrousavy/react-native-vision-camera/issues/1210 (comment))"

Sent me down the path of upgrading to RNVC v3, the latest version of React Native, the new react-native-worklets-core library, etc.

The result? My app builds successfully now, but there's no path forward for using Vision Camera Code Scanner. It just isn't going to happen, until someone fixes it to work with RNVC v3.

Unfortunately, there seem to be a grand total of 0 active contributors to the Vision Camera Code Scanner project -- on any repo. Hopefully this will change soon! I need it for a work project as well.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy