margelo/react-native-vision-camera #1693: ❓ Unable to get v3 working in project

willburstein, 2023-08-14:
### Question

I keep getting the error "Cannot read property 'installFrameProcessorBindings' of undefined, js engine: hermes" 

It also says:
[VisionCamera] node modules found at /Users/willburstein/Documents/Projects/Colog/node_modules/react-native-vision-camera/node_modules
[VisionCamera] react-native-worklets not found, Frame Processors disabled!
[VisionCamera] react-native-skia not found, Skia Frame Processors disabled!

on pod install.

### What I tried

I've applied the patch, double checked my versions, cleaned my build, reinstalled, and it only happens when I use the useCameraDevices() hook.

### VisionCamera Version

v3 r4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

DeveloperMCD, 2023-08-14:
I'm getting the same issue; I'm trying to use 3.0.0-rc.4

        "react-native": "0.72.3",
        "react-native-scan-barcode": "3.1.7",
        "react-native-reanimated": "3.4.2",
        "react-native-vision-camera": "3.0.0-rc.4",
        "vision-camera-code-scanner": "^0.2.0"

Here is the pertinent error:

> Project with path ':react-native-worklets' could not be found in project ':react-native-vision-camera'.

DeveloperMCD, 2023-08-14:
I commented out these two lines in /node_modules/react-native-vision-camera/android/build.gradle
// implementation project(":react-native-worklets")
//  implementation project(":shopify_react-native-skia")

Now I'm getting this error:
/home/matthew/dev/stackbayapp/node_modules/vision-camera-code-scanner/android/src/main/java/com/visioncameracodescanner/VisionCameraCodeScannerPluginPackage.java:19: error: cannot find symbol
    FrameProcessorPlugin.register(new VisionCameraCodeScannerPlugin());
                        ^
  symbol:   method register(VisionCameraCodeScannerPlugin)
  location: class FrameProcessorPlugin
4 errors

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':vision-camera-code-scanner:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

jthure, 2023-08-15:
[react-native-worklets](https://github.com/chrfalch/react-native-worklets) seems to be a required dependency. If you look at the package.json it's specified as a peerDependency. So you need to add that. [@shopify/react-native-skia](https://github.com/Shopify/react-native-skia) seems to be optional, so you should be able to make it work without, if you dont want to use the skia functionality

DeveloperMCD, 2023-08-15:
Now it gets further in the build process, dying here instead:

 [8/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [9/13] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPluginHostObject.cpp.o
  [10/13] Building CXX object CMakeFiles/VisionCamera.dir/home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o
  ninja: build stopped: subcommand failed.
  
  C++ build system [build] failed while executing:
      /home/matthew/programs/sdk/cmake/3.22.1/bin/ninja \
        -C \
        /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/.cxx/Debug/3f5ka3a3/arm64-v8a \
        VisionCamera
    from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android

DeveloperMCD, 2023-08-15:
One question -- should I follow someone else's advice (in another Github thread) and install it this way:
"react-native-worklets": "github:chrfalch/react-native-worklets#d62d76c",
or just do
npm i react-native-worklets

Which, at the time of this writing, yields
      "react-native-worklets": "^0.0.1-alpha",

DeveloperMCD, 2023-08-15:
:react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED |  
  | C/C++: ninja: Entering directory `/home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/.cxx/Debug/3f5ka3a3/arm64-v8a' |  
  | C/C++: In file included from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.cpp:5: |  
  | C/C++: /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.h:12:10: fatal error: 'react-native-worklets/WKTJsiWorklet.h' file not found |  
  | C/C++: /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/FrameHostObject.cpp:11:10: fatal error: 'react-native-worklets/WKTJsiHostObject.h' file not found |  
  | C/C++: In file included from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/VisionCameraProxy.cpp:5: |  
  | C/C++: In file included from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/VisionCameraProxy.h:10: |  
  | C/C++: /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JVisionCameraProxy.h:9:10: fatal error: 'react-native-worklets/WKTJsiWorkletContext.h' file not found |  
  | C/C++: In file included from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/VisionCamera.cpp:4: |  
  | C/C++: /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.h:12:10: fatal error: 'react-native-worklets/WKTJsiWorklet.h' file not found |  
  | C/C++: In file included from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/JSIJNIConversion.cpp:5: |  
  | C/C++: In file included from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/JSIJNIConversion.h:7: |  
  | C/C++: In file included from /home/matthew/.gradle/caches/transforms-3/9bdaeac874ab8f2ffcc97e83c7b54687/transformed/jetified-react-android-0.72.4-debug/prefab/modules/jsi/include/jsi/jsi.h:13: |  
  | C/C++: In file included from /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/functional:504: |  
  | C/C++: /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:4434:5: error: static_assert failed due to requirement 'is_constructible<vision::FrameHostObject, facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>::value' "Can't construct object in make_shared" |  
  | C/C++: /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/JSIJNIConversion.cpp:120:28: note: in instantiation of function template specialization 'std::make_shared<vision::FrameHostObject, facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>' requested here |  
  | C/C++: In file included from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/JSIJNIConversion.cpp:5: |  
  | C/C++: In file included from /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/JSIJNIConversion.h:7: |  
  | C/C++: In file included from /home/matthew/.gradle/caches/transforms-3/9bdaeac874ab8f2ffcc97e83c7b54687/transformed/jetified-react-android-0.72.4-debug/prefab/modules/jsi/include/jsi/jsi.h:13: |  
  | C/C++: In file included from /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/functional:504: |  
  | C/C++: /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:2278:9: error: field of type 'vision::FrameHostObject' has private constructor |  
  | C/C++: /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:2362:9: note: in instantiation of function template specialization 'std::__compressed_pair_elem<vision::FrameHostObject, 1, false>::__compressed_pair_elem<facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &, 0>' requested here |  
  | C/C++: /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:3635:16: note: in instantiation of function template specialization 'std::__compressed_pair<std::allocator<vision::FrameHostObject>, vision::FrameHostObject>::__compressed_pair<std::allocator<vision::FrameHostObject> &, facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>' requested here |  
  | C/C++: /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:4441:26: note: in instantiation of function template specialization 'std::__shared_ptr_emplace<vision::FrameHostObject, std::allocator<vision::FrameHostObject>>::__shared_ptr_emplace<facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>' requested here |  
  | C/C++: /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/JSIJNIConversion.cpp:120:28: note: in instantiation of function template specialization 'std::make_shared<vision::FrameHostObject, facebook::jni::basic_strong_ref<vision::JFrame, facebook::jni::LocalReferenceAllocator> &>' requested here |  
  | C/C++: /home/matthew/dev/rn/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/FrameHostObject.h:21:12: note: implicitly declared private here

<br class="Apple-interchange-newline">

jthure, 2023-08-15:
> One question -- should I follow someone else's advice (in another Github thread) and install it this way: "react-native-worklets": "github:chrfalch/react-native-worklets#d62d76c", or just do npm i [...]

The version published on npm seems to be very old. I looked in the example project of this repo, there you can see that the github repo is used as dep, so I would go with that. Although it would be nice if an up to date version of react-native-worklets got published to npm

jthure, 2023-08-15:
> :react-native-vision-camera:buildCMakeDebug[arm64-v8a] FAILED |     | C/C++: ninja: Entering directory [...]

The FrameHostObject has a private constructor (a mistake), it needs to be patched in the rc4. Look for the public modifier which is applied to the destructor, but not the constructor, and just move the constructor into public modifier scope as well

DeveloperMCD, 2023-08-15:
Thank you for the help @jthure! I'll try that.

mrousavy, 2023-08-16:
I think the JS codebase isn't realizign that RN Worklets is optional yet. Only native knows that.

Will fix soon

jthure, 2023-08-16:
> I think the JS codebase isn't realizign that RN Worklets is optional yet. Only native knows that.
> 
> Will fix soon

Yea, I noticed that. I tried taking a stab at it myself, but wasn't able to find a way to do it. Same goes for skia as well, ts expects some skia types to exist. If you have an idea on how to do it and want some help, let me know.

mrousavy, 2023-08-16:
Well I think it's because `index.ts` imports and re-exports all files, even the FrameProcessorPlugins and useFrameProcessor files. 

I'm not sure how to make those optional tbh... Something like a `try { require(..) } catch { }` breaks babel, right?

sergey-shablenko, 2023-08-16:
Hi, how do I fix this error

```c++
error: no member named 'alias_refJFrame' in namespace 'facebook::jni'
```

I have tried to apply this hotfix
```c++
class JSI_EXPORT FrameHostObject : public jsi::HostObject {
  public:
  FrameHostObject(const jni::alias_refJFrame::javaobject& frame);
  ~FrameHostObject();
```

mrousavy, 2023-08-17:
There are missing `<` and `>` around the `JFrame::javaobject` in that code huh

DeveloperMCD, 2023-08-17:
"[@shopify/react-native-skia](https://github.com/Shopify/react-native-skia) seems to be optional, so you should be able to make it work without, if you dont want to use the skia functionality"

If you don't mind my asking, what exactly is "skia functionality"? What service does skia provide this library?
I do need FrameProcessors, since one of my use-cases is barcode scanning.

marouane-edghoughi, 2023-08-21:
Currently having the same issue, did you find any workaround? @mrousavy

Ivanindiani, 2023-08-22:
> there is another error
> 
> ```c++
> fatal error: 'react-native-worklets/WKTJsiWorkletContext.h' file not found
> #include <react-native-worklets/WKTJsiWorkletContext.h>
> ```
> 
> I have installed worklets and skia, not sure why it does not see it. Is there any chance of getting a working version any time soon?

+1

bglgwyng, 2023-08-23:
I believe I've identified the issue.
 It appears that there's a problem with circular dependencies between [Camera.tsx](https://github.com/mrousavy/react-native-vision-camera/blob/v3/src/Camera.tsx) and [FrameProcessorPlugins.ts](https://github.com/mrousavy/react-native-vision-camera/blob/v3/src/FrameProcessorPlugins.ts) within the codebase.
It seems that in certain Babel configurations, such circular dependencies might be resolved correctly. 
However, people in this issue might not benefit from those configurations. 
In my case, I'm using Expo, which employs its own Babel setup, and it seems that this is causing the observed behavior.

Here's a quick fix: https://gist.github.com/bglgwyng/8f13c9c82eb900ade5d8b7e83245fb21. 
In my perspective, it's not necessary for `installFrameProcessorBindings` to be a static method within the Camera module. So, it would be considerable to directly use `CameraModule.installFrameProcessorBindings` to address the circular dependency.

bglgwyng, 2023-08-23:
@mrousavy, I attempted to create a PR, but just came back as I realized there is more than one way to resolve this issue. I suppose that you have your own way of reorganizing files with circular dependencies. So, just noticing you would be enough. Thanks!

DeveloperMCD, 2023-08-23:
Just so you know, that is NOT the only issue. I'm not using Expo, so that can't be the source of the problem in my case.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀

Maheshbhusanoor, 2024-01-29:
Does anybody know the fix for the following error?

"react-native": "0.72.5",
"react-native-reanimated": "3.3.0",
 "react-native-vision-camera": "3.7.1",
 "react-native-worklets-core": "^0.2.0",

> Task :app:compileDevelopmentDebugJavaWithJavac
/Users/maheshbhusanoor/MCOM_HM/MPOS/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPluginPackage.java:19: error: cannot find symbol
    FrameProcessorPlugin.register(new VisionCameraCodeScannerPlugin());
                        ^
  symbol:   method register(VisionCameraCodeScannerPlugin)
  location: class FrameProcessorPlugin
/Users/maheshbhusanoor/MCOM_HM/MPOS/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:40: error: VisionCameraCodeScannerPlugin is not abstract and does not override abstract method callback(Frame,Map<String,Object>) in FrameProcessorPlugin
public class VisionCameraCodeScannerPlugin extends FrameProcessorPlugin {
       ^
/Users/maheshbhusanoor/MCOM_HM/MPOS/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:62: error: method does not override or implement a method from a supertype
  @Override
  ^
/Users/maheshbhusanoor/MCOM_HM/MPOS/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:251: error: incompatible types: String cannot be converted to Map<String,Object>
    super("scanCodes");
          ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
4 errors

PR #1759 (2023-09-18): feat: Reanimated v3 support in RNVisionCamera v2
## What

**This PR makes it possible to use RNVisionCamera v2 with Reanimated v3 (3.5.0+)** 🥳 

I've branched off from @mrousavy's PR (https://github.com/mrousavy/react-native-vision-camera/pull/1623) which bumps react-native version in the example app as well as does some general cleanup and integrated Reanimated v3 on top of it.

## Changes

* Move Camera thread runtime creation from C++ to JS using `Reanimated.createWorkletRuntime`
* Call worklets using `WorkletRuntime::runGuarded` which already supports error handling
* Consume Reanimated as [... cut]