margelo/react-native-vision-camera #1696: 🐛 Unable to build v3 in iOS

DeveloperMCD, 2023-08-16:
### What were you trying to do?

Trying to successfully build project with v3.0.0-rc4 on iOS platform.

### Reproduceable Code

```tsx
npx react-native run-ins
```

### What happened instead?

** BUILD FAILED **

The following build commands failed:
	CompileC /Users/developer/Library/Developer/Xcode/DerivedData/stackbay-hkdvgolugzmnjnbykqfllfpapwnp/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/SkImageHelpers.o /Users/developer/Documents/ios/stackbayapp/node_modules/react-native-vision-camera/ios/Skia\ Render\ Layer/SkImageHelpers.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
(1 failure)

### Device

iPhone 13

### VisionCamera Version

3.0.0-rc4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DeveloperMCD, 2023-08-22:
Just got the latest version 3.0.0-rc5 and updated the pod file to use "react-native-worklets-core" version 0.2.0

Still getting an error on build, and still under "skia" --

The following build commands failed:
	CompileC /Users/developer/Library/Developer/Xcode/DerivedData/stackbay-hkdvgolugzmnjnbykqfllfpapwnp/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/SkImageHelpers.o /Users/developer/Documents/ios/stackbayapp/node_modules/react-native-vision-camera/ios/Skia\ Render\ Layer/SkImageHelpers.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
(1 failure)

patras1, 2023-08-23:
Having a similar issue:
[!] Unable to find a specification for `react-native-worklets` depended upon by `VisionCamera`

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

I did install the 'react-native-worklets-core' .
Removed the old `react-native-worklets` from package-lock.json
Update to the latest version 3.0.0-rc5.
Run pod install 

what am I missing?
Thanks

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy