margelo/react-native-vision-camera #1697: 🐛 frame-processor/unavailable: Frame Processors are not enabled

ademirtemur, 2023-08-17:
### What were you trying to do?

I am trying to scan qr code, camera is running when using without frameprocessor but when set frameprocessor i am getting this error. 
frame-processor/unavailable: Frame Processors are not enabled. See https://react-native-vision-camera.com/docs/guides/troubleshooting

"react-native": "0.72.3"
"react-native-reanimated": "^3.3.0"
"react-native-vision-camera": "^2.15.6",
"vision-camera-code-scanner": "^0.2.0"

import "react-native-reanimated"

babel.config.js
plugins: [
    [
      'react-native-reanimated/plugin',
      {
        relativeSourceLocation: true,
        globals: ['__scanCodes'],
      }
    ]
  ]

const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.ALL_FORMATS], { checkInverted: false })

<Camera
    style={[{ flex: 1 }, StyleSheet.absoluteFill]}
    device={device}
    isActive={true}
    frameProcessorFps={5}
    frameProcessor={frameProcessor}
/>

### What happened instead?

camera is running when using without frameprocessor but when set frameprocessor i am getting this error. 

frame-processor/unavailable: Frame Processors are not enabled. See https://react-native-vision-camera.com/docs/guides/troubleshooting

### Device

android and ios

### VisionCamera Version

2.15.6

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ademirtemur, 2023-08-17:
@mrousavy

BH-Mobile, 2023-08-18:
Have you found a solution? I have android works, but IOS does not work

mrousavy, 2023-08-18:
Not enough details (you did not put any log output here), closing this issue.

closed 2023-08-18 by mrousavy

BH-Mobile, 2023-08-21:
Solution for IOS.
in PodFile:
pre_install do |installer|
    installer.pod_targets.each do |pod|
      if pod.name.eql?('vision-camera-code-scanner') || pod.name.eql?('VisionCamera') ||  pod.name.eql?('RNReanimated')
        def pod.build_type
          Pod::BuildType.static_library
        end
      end
    end
  end

Gozgur, 2023-08-21:
What to fix this issue on Android? I have same versions.

carlgrob5171989, 2023-10-02:
What is the resolution? the issue is occurring on both android and ios .

pgpankajcapgemini, 2023-10-03:
I am also facing same issue for iOS.

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.