margelo/react-native-vision-camera #1698: 🐛rn 0.72.3   Property '_setGlobalConsole' doesn't exist

ademirtemur, 2023-08-17:
### What were you trying to do?

ReferenceError: Property '_setGlobalConsole' doesn't exist

node 18.15.0

"react-native": "0.72.3"
"react-native-reanimated": "^3.3.0"
"react-native-vision-camera": "^2.15.6"

### What happened instead?

ReferenceError: Property '_setGlobalConsole' doesn't exist

### Device

android

### VisionCamera Version

2.15.6

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ademirtemur, 2023-08-17:
@mrousavy

mrousavy, 2023-08-18:
Not enough information.

closed 2023-09-01 by mrousavy