margelo/react-native-vision-camera #1699: Camera with frame processor needs to be specific dimensions or it break

lninnn, 2023-08-17:
### What were you trying to do?

Create a camera view component with a left tab and a right tab

### Reproduceable Code

```tsx
<View style={{flex: 1}}>
      <CameraLeftIcons/> 
//common view with button
</View>
<View style={{ flex: 5 }}>
     <Camera
        ref={camera}
        device={device}
        isActive={true}
        photo={true}
        frameProcessorFps={5}
        frameProcessor={frameProcessor}
     />
<View
style={{flex: 1}}>
      <CameraRightIcons/>
</View>
```

### What happened instead?

The app crashes when camera is not full screen or it is not a specific ratio only when frame processor is on. With frame processor off, the camera flex fully to the size it was given

### Device

Android Galaxy 13

### VisionCamera Version

2.15.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-08-18:
Hey this might be fixed in VisionCamera V3 I think

mrousavy, 2023-09-05:
@akshay8033 create a separate issue.

closed 2023-09-05 by mrousavy