margelo/react-native-vision-camera #1704: 🐛 3.0.0-rc5 uses new "react-native-worklets" library, now called "react-native-worklets-core"

DeveloperMCD, 2023-08-22:
### What were you trying to do?

The library is on a different github account and everything.
https://github.com/margelo/react-native-worklets-core

The good news? I can use the very latest 3.0.0-rc5
The bad news? I had to go through and change some files to use react-native-worklets-core on the new github acct.

        "react-native-vision-camera": "3.0.0-rc.5",
        "react-native-worklets-core": "github:margelo/react-native-worklets-core#f0e893e",

I went through about 7 files and made changes like this:

#include <react-native-worklets-core/WKTJsiWorklet.h>
#include <react-native-worklets-core/WKTJsiHostObject.h>

And now I'm happily building the latest reanimated, vision camera, react native, etc.

Problem is, I had to edit some files, and I can't commit them to the repo. So this needs to be fixed for real. Either I need to "alias" the library as "react-native-worklets", or it needs to be renamed, or something.

### What happened instead?

Got a lot of errors where it couldn't find "react-native-worklets"

### Device

Android 12

### VisionCamera Version

3.0.0-rc5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-08-22:
Hi!

I already said this here: https://github.com/mrousavy/react-native-vision-camera/issues/1376#issuecomment-1686113825

V3 uses RN Worklets Core now. From npm. No more GitHub

closed 2023-08-22 by mrousavy

patras1, 2023-08-23:
How do I remove the old 'react-native-worklets`?
Because I'm getting:
[!] Unable to find a specification for `react-native-worklets` depended upon by `VisionCamera`

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

Althow I removed it from package-lock.json, run npm install and pod install