margelo/react-native-vision-camera #1708: 🐛 undefined symbol: RNWorklet::JsiWrapper

DeveloperMCD, 2023-08-22:
### What were you trying to do?

Trying to build my project again -- it worked this morning! 
All I did was try to use vision-camera-code-scanner with 3.0.0-rc3, and now I can't go back. I've purged all my files, including node_modules, npm cache, gradle, and android build folder. 

Here are the relevant libraries that make any difference:
        "@shopify/react-native-skia": "^0.1.202",
        "react": "18.2.0",
        "react-native": "0.72.4",
        "react-native-reanimated": "3.4.2",
        "react-native-vision-camera": "3.0.0-rc.5",
        "react-native-worklets-core": "^0.2.0",

### Reproduceable Code

```tsx
npx react-native start
```

### What happened instead?

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/.cxx/Debug/j6t3e341/arm64-v8a'

### Relevant log output

```shell
[12/14] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JVisionCameraProxy.cpp.o
  [13/14] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o
  [14/14] Linking CXX shared library ../../../../build/intermediates/cxx/Debug/j6t3e341/obj/arm64-v8a/libVisionCamera.so
  FAILED: ../../../../build/intermediates/cxx/Debug/j6t3e341/obj/arm64-v8a/libVisionCamera.so 
  : && /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -DSK_GL -DSK_GANESH -DSK_BUILD_FOR_ANDROID -g  -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Qunused-arguments -Wl,--no-undefined -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/j6t3e341/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/skia/SkiaRenderer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JVisionCameraScheduler.cpp.o  /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/aarch64-linux-android/21/liblog.so  -landroid  /home/matthew/.gradle/caches/transforms-3/ed5971bff14b029db9d61e0025901397/transformed/jetified-react-android-0.72.4-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /home/matthew/.gradle/caches/transforms-3/ed5971bff14b029db9d61e0025901397/transformed/jetified-react-android-0.72.4-debug/prefab/modules/reactnativejni/libs/android.arm64-v8a/libreactnativejni.so  /home/matthew/.gradle/caches/transforms-3/ed5971bff14b029db9d61e0025901397/transformed/jetified-react-android-0.72.4-debug/prefab/modules/folly_runtime/libs/android.arm64-v8a/libfolly_runtime.so  /home/matthew/.gradle/caches/transforms-3/d2945effe2c76af5abf2618fac389cac/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  -lGLESv2  -lEGL  -ljnigraphics  /home/matthew/dev/stackbayapp/node_modules/@shopify/react-native-skia/libs/android/arm64-v8a/libskia.a  /home/matthew/dev/stackbayapp/node_modules/@shopify/react-native-skia/libs/android/arm64-v8a/libsvg.a  /home/matthew/dev/stackbayapp/node_modules/@shopify/react-native-skia/libs/android/arm64-v8a/libskshaper.a   -latomic -lm && :
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
  >>> referenced by VisionCameraProxy.cpp:35 (/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/VisionCameraProxy.cpp:35)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(vision::VisionCameraProxy::~VisionCameraProxy())
  >>> referenced by JFrameProcessor.cpp:39 (/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/java-bindings/JFrameProcessor.cpp:39)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::JsiHostObject()
  >>> referenced by WKTJsiWorklet.h:77 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:77)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(RNWorklet::JsiWorklet::JsiWorklet(facebook::jsi::Runtime&, facebook::jsi::Value const&))
  >>> referenced by WKTJsiWorkletContext.h:31 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorkletContext.h:31)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::~JsiHostObject()
  >>> referenced by WKTJsiWorklet.h:79 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:79)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(RNWorklet::JsiWorklet::JsiWorklet(facebook::jsi::Runtime&, facebook::jsi::Value const&))
  >>> referenced by WKTJsiWorklet.h:74 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:74)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(RNWorklet::JsiWorklet::~JsiWorklet())
  
  ld: error: undefined symbol: RNWorklet::JsiWrapper::wrap(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWrapper*)
  >>> referenced by WKTJsiWrapper.h:62 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWrapper.h:62)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(RNWorklet::JsiWorklet::createWorklet(facebook::jsi::Runtime&, std::__ndk1::shared_ptr<facebook::jsi::Function>))
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::get(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&)
  >>> referenced by VisionCameraProxy.cpp
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(vtable for RNWorklet::JsiWorklet)
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::set(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&, facebook::jsi::Value const&)
  >>> referenced by VisionCameraProxy.cpp
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(vtable for RNWorklet::JsiWorklet)
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::getPropertyNames(facebook::jsi::Runtime&)
  >>> referenced by VisionCameraProxy.cpp
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraProxy.cpp.o:(vtable for RNWorklet::JsiWorklet)
  
  ld: error: undefined symbol: typeinfo for RNWorklet::JsiHostObject
[... 41 lines cut ...]
  clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
  ninja: build stopped: subcommand failed.
  
  C++ build system [build] failed while executing:
      /home/matthew/programs/sdk/cmake/3.22.1/bin/ninja \
        -C \
        /home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/.cxx/Debug/j6t3e341/arm64-v8a \
[... cut ...]
### Device

Android 12

### VisionCamera Version

3.0.0-rc5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

tyanbiao, 2023-08-23:
same error!

DeveloperMCD, 2023-08-23:
I found the problem!
Frame Processors are *never* being enabled, they can't be, so they're not being linked in.

In /android/CMakeLists.txt:

set(CMAKE_VERBOSE_MAKEFILE ON)
set(PACKAGE_NAME "VisionCamera")
set(BUILD_DIR ${CMAKE_SOURCE_DIR}/build)
set(CMAKE_VERBOSE_MAKEFILE ON)
set(ENABLE_FRAME_PROCESSORS ON)    <-------------------Need to add this line
set(CMAKE_CXX_STANDARD 17)

Fabianurrutia, 2023-09-04:
If someone continues to have this problem with version 3.0.0, I recommend the following:
Clone the original project from the main branch
go to the package folder and do npm i -> npm run build
copy the lib folder and paste it into the folder
node_modules/react-native-vision-camera/
run again

This will update the version that is in main that is not yet as a release candidate, nor as a release version.

(To keep these changes I use patch package)
npx patch-package react-native-vision-camera
and follow the steps in the patch-package package

**FAST STEP:**
If you want only the changes of patch-package copy this on your patches folder. (See the patch-package docs)
[react-native-vision-camera+3.0.0.patch](https://github.com/mrousavy/react-native-vision-camera/files/12515971/react-native-vision-camera%2B3.0.0.patch)

stemyke, 2023-09-05:
@Fabianurrutia I got errors like this:

ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()

ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnJsThread(std::__ndk1::function<void (facebook::jsi::Runtime&)>&&)

ld: error: undefined symbol: RNWorklet::JsiWorkletContext::runtimeMappings

ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__ndk1::function<void (RNWorklet::JsiWorkletContext*, facebook::jsi::Runtime&)>&&)

ld: error: undefined symbol: RNWorklet::JsiWorkletContext::defaultInstance

ld: error: undefined symbol: RNWorklet::JsiHostObject::JsiHostObject()

ld: error: undefined symbol: vtable for RNWorklet::JsiWorkletContext

ld: error: undefined symbol: RNWorklet::JsiWorkletContext::~JsiWorkletContext()

ld: error: undefined symbol: RNWorklet::JsiWorkletContext::JsiWorkletContext(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&, facebook::jsi::Runtime*, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>, std::__ndk1::function<void (std::__ndk1::function<void ()>&&)>)

ld: error: undefined symbol: RNWorklet::JsiHostObject::~JsiHostObject()

ld: error: undefined symbol: RNWorklet::JsiWrapper::wrap(facebook::jsi::Runtime&, facebook::jsi::Value const&, RNWorklet::JsiWrapper*)

ld: error: undefined symbol: RNWorklet::JsiHostObject::get(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&)

ld: error: undefined symbol: RNWorklet::JsiHostObject::set(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&, facebook::jsi::Value const&)

ld: error: undefined symbol: RNWorklet::JsiHostObject::getPropertyNames(facebook::jsi::Runtime&)

ld: error: undefined symbol: typeinfo for RNWorklet::JsiHostObject

Doy ou have any idea?

metrix-hu, 2023-09-05:
@mrousavy Do you have any idea of what this could be? It is still a problem with 3.0.0

mrousavy, 2023-09-05:
this is not a VisionCamera issue but a worklets issue.
Clean yo cache

closed 2023-09-05 by mrousavy

metrix-hu, 2023-09-05:
@mrousavy What cache should I clear exactly? I tried this in a from scratch project also and have the same issue there.

mrousavy, 2023-09-05:
hm can you reproduce it in the example app? make sure your environment is up to date with node, android studio, homebrew etc

metrix-hu, 2023-09-05:
Can it be a problem with RN 0.72 (I use it in the new projet) or should it be compatible with it also?

metrix-hu, 2023-09-05:
I use these versions in my build..gradle

ext {
        buildToolsVersion = "33.0.0"
        minSdkVersion = 21
        compileSdkVersion = 33
        targetSdkVersion = 33

        // We use NDK 23 which has both M1 support and is the side-by-side NDK version from AGP.
        ndkVersion = "23.1.7779620"
    }
    
   Android studio and node is also up-to-date. I am on Windows 11.

metrix-hu, 2023-09-05:
The problem for me was the minSdkVersion. I set it to 26 as in the example app and it seems to be working now.

bglgwyng, 2023-09-05:
I've encountered the same issue in the past, but I was unable to determine a reliable way to replicate this build error. 
It appears that simply clearing the cache and reinstalling everything may have resolved the problem, although I cannot be 100% certain.
Could someone share the repository that triggers this error? I don't have one now.

bglgwyng, 2023-09-05:
@metrix-hu Fortunately(?), I just faced this problem again and your solution(set `minSdkVersion` to 26) fixed my problem. Thanks! However, I'm not sure if it's a truly valid solution so that the same error never happens again under the current configuration. 
Could you offer some rationale behind your proposed solution?

metrix-hu, 2023-09-05:
@bglgwyng As I said before I scaffolded a brand new RN 0.72 project installed the needed packages and I instantly faced the error. As Marc suggested I tried running the Example app also with the exact same configuration I had before and it worked like a charm. Every dependency was updated and was still not working. So node was on latest 18.x version Androis Studio is latest possible everything. My project was still not working and the example app worked. So I tried to find what the heck is the difference between the two, and this was the only thing I found. I changed it after that it instantly worked in my project also.

mrousavy, 2023-09-05:
Ohhh, that's weird. Yea maybe it's because of minSdk, that **has to be 26** for VisionCamera at least. Interesting catch guys. It is stated in the troubleshooting docs btw

metrix-hu, 2023-09-05:
@mrousavy No way I am struggling with reanimated v3 couldnt make it because of a typescript error. Now I tried theirt latest nightly which needed a new native build, and now I got thies error again..

metrix-hu, 2023-09-06:
@mrousavy What the heck. Today I reproduces the issue in the example app too.

[image]

I did not change anything I just hit "build/clean project" in Android Studio before a new build, otherwise it would just start the app instantly again.

mrousavy, 2023-09-06:
Maybe this is a Windows issue. I don't know. Sorry to ping you here, but @chrfalch - any thoughts?

metrix-hu, 2023-09-06:
@mrousavy Maybe worklets core does some copying of files when building C++? I can see in Android Studio that the compilation of worklets-core and vision-camera happens in parallel, maybe that causes the problem?

chrfalch, 2023-09-06:
Sounds really strange - would it be possible to test adding only the worklets library in an empty project @metrix-hu and see if you had any issues?

The library is found here: https://github.com/margelo/react-native-worklets-core.

A great test would even be to test its example app in your environment.

The only thing I could come up with in regards to Windows would have to be that referencing paths in CMakeLists.txt might be odd somehow?

metrix-hu, 2023-09-06:
@chrfalch I tried the example app from react-native-worklets-core. After yarn installing it I opened the android folder inside the example folder, and here gradle sync failed with the following:

[image]

metrix-hu, 2023-09-06:
@chrfalch You know what is also strange? 

1. I tried to scaffold a new project added worklets-core as dependency. Everything is worked like a charm here. Tried cleaning the project to ruin it and build it again it was still working.
2. I added vision-camera also as a dependency. Tried to build the project but I run into the same error. Now I run the cleaning again and it solved the problem..
3. After I tried to run the clean commands again in the main project too and strangely it built here again..

I still think that something is conflicting here that the two build processes are running in parallel, and now I just got lucky that worklets-core finished before than vision-camera needed the files..

mrousavy, 2023-09-06:
@chrfalch this is how I get `node_modules` in VisionCamera: 

https://github.com/mrousavy/react-native-vision-camera/blob/eddb01fda1cd4214a6159aba93133564c3220c55/package/android/build.gradle#L51-L64

This is how we do it in Worklets Core:

https://github.com/margelo/react-native-worklets-core/blob/94eaf5d828118510eae1ec16876ec384f856ebe2/android/build.gradle#L15-L28

Exactly the same lol. Maybe somewhere there's a `/`, which would not be working on Windows? Idk, I can take a look at this when I'm back home

bglgwyng, 2023-09-06:
It might not be Windows only issue. I'm building my project on a Mac, but I'm encountering the same error.

DeveloperMCD, 2023-09-06:
It is *not* a Windows issue. I only use Linux.

DeveloperMCD, 2023-09-06:
I'm going to be "that guy" who says I fixed it, and hopefully not too many people reading this will say "I TRIED THAT -- AND I'M STILL HAVING THE PROBLEM!" haha
Here is the fix:
Go in to android/CMakeLists.txt and add a line at the top, so it looks like this:

set(CMAKE_VERBOSE_MAKEFILE ON)
set(PACKAGE_NAME "VisionCamera")
set(BUILD_DIR ${CMAKE_SOURCE_DIR}/build)
set(CMAKE_VERBOSE_MAKEFILE ON)
set(ENABLE_FRAME_PROCESSORS ON)
set(CMAKE_CXX_STANDARD 17)

You should then be able to build. Those errors above are clearly showing a failure to link Worklets Core in particular. This actually makes sense as a solution.

metrix-hu, 2023-09-06:
@DeveloperMCD It looks like that line is already there after BUILD_DIR

metrix-hu, 2023-09-06:
Oh you may talk about the ENABLE_FRAME_PROCESSORS one. I am not sure that is correct because it should be calculated based on if worklets-core is existing or not, right?

DeveloperMCD, 2023-09-06:
> Oh you may talk about the ENABLE_FRAME_PROCESSORS one. I am not sure that is correct because it should be calculated based on if worklets-core is existing or not, right?

It seems to *never* be calculated. Please show me where the variable "ENABLE_FRAME_PROCESSORS" is set. 
I'll wait :)

metrix-hu, 2023-09-06:
@DeveloperMCD You dont have to be so rude because someone is assuming based on rational things. How do you explain the reason of this variable existing if it always needs to have the value "ON" to make it build? But look I searched it for you.

[image]

It should come from the build.gradle file as an argument. , where the value is based on the `hasWorklets` variable which is calculated by the worklets library existence. I am not really familiar how CMakeLists or things like this work exactly so lets ask @mrousavy  again he is a lot smarter than me. The build script logs a lot than Frame Processors are enabled even without this line, there should be some reason for it. If this constant line is the solution, then why is it so random that it sometimes work and sometimes it doesnt?

DeveloperMCD, 2023-09-06:
I wasn't being rude. You're having a misunderstanding based on differing cultures/generations. Words don't always come across as the typer intended them. It's a classic flaw of text-based communication. I'll leave it at that.

DeveloperMCD, 2023-09-06:
"You dont have to be so rude because someone is assuming based on rational things. How do you explain the reason of this variable existing if it always needs to have the value "ON" to make it build?"

It's simple. MRousavy might have made a mistake. A careless error. I consider it possible that he wrote 99 lines of perfect code but forgot the first line of code to set the whole ball rolling. It's happened before to me. We all make mistakes. 

Which is why I don't assume anything going in to a troubleshooting adventure. You have to start with zero assumptions, verifying everything as you go. Eventually you find the problem.

metrix-hu, 2023-09-06:
@DeveloperMCD Yes maybe I believe you but you still did not explain anything. I am not assuming that anybody is perfect here, you are misunderstanding me also. I am assuming that it should not be always ON because then why anyone would introduce it in the first place? However I tried your suggestion and I am able to build (however I was able to do it last time even without this change..) but now I got a runtime error, that Frame Processors are not enabled! Even if in the build logs it still says that they are enabled... How can you explain that?

[image]

[image]

DeveloperMCD, 2023-09-06:
Just for starters, I would run this code, or something similar to it for your operating system:
cd <project-folder>
rm -rf node_modules
rm -rf android/build
rm -f package-lock.json
rm -rf ~/.gradle
rm -rf ~/.npm

This would allow for a complete reset of the build process.

metrix-hu, 2023-09-06:
What if I use yarn? And why would I remove lock files if we are not changing any package version here?

DeveloperMCD, 2023-09-06:
Then do a full reset as per your operating system and particular build chain.

Also what does your Babel config look like?
Mine is:

module.exports = {
    presets: ['module:metro-react-native-babel-preset'],
    plugins: [
        [
            'react-native-reanimated/plugin',
            {
                globals: ['__scanCodes'],
            },
        ],
        ['react-native-worklets-core/plugin'],
    ],
};

I can't give you a full reason or explanation for what's going on with the Makefile. That will have to wait for MRousavy. I just know how I fixed it for my own particular case.

DeveloperMCD, 2023-09-06:
One problem -- you need to get rid of Skia. MRousavy got rid of it completely when he released 3.0.0 a few days ago.
I have it Skia disabled (plus absent) in my own project.

VisionCamera_disableSkia=true
VisionCamera_buildToolsVersion=33.0.0
VisionCamera_compileSdkVersion=33
VisionCamera_kotlinVersion=1.7.00
VisionCamera_targetSdkVersion=33
VisionCamera_ndkVersion=23.1.7779620
android.enableJetifier=true
android.useAndroidX=true

metrix-hu, 2023-09-07:
@DeveloperMCD Thanks man for the effort and ideas you gave me. Sorry for my odd comments from yesterday I was so stressed out about this.. Now I started everything from scratch tried every combination, but unfortunately I still have the same problem as in my last comment. The build only works for me when I does not change anything and I got lucky in that moment and packages build in the correct order. Its so weird. 

But in this case I still have an other runtime error..: https://github.com/mrousavy/react-native-vision-camera/issues/1776

metrix-hu, 2023-10-10:
@KraljSamo I tried with the latest version of VisionCamera v3 a few days ago and now everything works well so far with the frameprocessors.

Muhammad-Ahmad8, 2024-10-21:
> I'm going to be "that guy" who says I fixed it, and hopefully not too many people reading this will say "I TRIED THAT -- AND I'M STILL HAVING THE PROBLEM!" haha Here is the fix: Go in to node_modules/react-native-vision-camera/android/CMakeLists.txt and add a line at the top, so it looks like this:
> 
> set(CMAKE_VERBOSE_MAKEFILE ON) set(PACKAGE_NAME "VisionCamera") set(BUILD_DIR ${CMAKE_SOURCE_DIR}/build) set(CMAKE_VERBOSE_MAKEFILE ON) set(ENABLE_FRAME_PROCESSORS ON) <----------- THIS LINE IS ADDED set(CMAKE_CXX_STANDARD 17)
> 
> You should then be able to build. Those errors above are clearly showing a failure to link Worklets Core in particular. This actually makes sense as a solution.

This fixed my issue. Thanks

i6mi6, 2024-12-12:
Leaving it here for my future self - what works consistently is **first** Build -> Rebuild Project from Android studio, and **only then** running or making a build

stemyke, 2024-12-13:
> Leaving it here for my future self - what works consistently is **first** Build -> Rebuild Project from Android studio, and **only then** running or making a build

Thats exactly how it works for me too, even with v4. And for production build I also have to do a complete reinstall of node modules after deleting them.