margelo/react-native-vision-camera #1709: ❓ Restart Vision camera after using local stream

RaylissW, 2023-08-23:
### Question

Well, basically i try to share photos and local stream between devices. I use _react-native-webrtc_ to share localStream, sometimes remote user requires a photo, so i use _react-native-vision-camera_ to do so.

It works fine until you try to make the second photo. It seems like there is no way to restart vision camera after resetting the local stream.

Here is my camera component:
`   {(cameraDisplay ) && (
          <ViewShot  >
                    <Camera
                      ref={camera}
                      style={StyleSheet.absoluteFill}
                      device={device}
                      photo={true}
                      onInitialized={onCameraInitialized}
                      isActive={cameraDisplay}
                    />
          </ViewShot>
     )}`
                    
when remote user requires a photo, i do this:
` case 'NEED_PHOTO': {
       localStream.current.getVideoTracks().forEach(track => track.enabled = false);
       setTimeout(()=>{
       if (appState.isNeedPhotoAnswer){
              return null
       }
       **setCameraDisplay(true);**
       },5000);
       setAppState(a=>({...a, isNeedPhotoAnswer: true}));
       return;
    }`
    
    I tried to disable videoTrack for Vision camera to capture current. It did not work, so i tried to give it sometime to shut down with setTimeout (it went with an error [Cannot read property 'apply' of undefined, js engine: hermes], then i added some state to avoid dublication. Then i FINALLY  **setCameraDisplay(true);** TRUE to mount a camera component.
    
    ` const onCameraInitialized = useCallback(() => {
          setIsCameraInitialized(true);
  }, []);

useEffect(() => {
if(camera && camera.current && isCameraInitialized){
   snapPhoto();
}
},[isCameraInitialized])`

Then i wait for camera to initialize and wait again before making photo. I guess now it is a little overcomplicated, but i just could not understand whether it mounts or not. The camera display appears only on the first time of using this function. On the second time nothing is displayed. 

Is there another way to mount camera? I would really appreciate any help on my problem!

btw after getting an image, remote user may close it, then i restore state:

`
 case 'FILE_CLOSED':{
      console.warn(calleeId.current);
      setCameraDisplay(false);
      setIsCameraInitialized(false);
      //stopConversation({needNotifyRemoteCaller: false, needStopLocalStream: true, clearCalleeId: false,  calleeId: calleeId.current});
      restartLocalStream();
      return  setAppState(a => ({...a, viewedObject: "", viewedObjectName:  '' , viewedObjectType:  '', isNeedPhotoAnswer: false }));
    }
`

+ restarting call does not make any effect either. Unability to use camera REMAINS after reload and there is no way of getting picture until reinstalling the app.
i am a new react Native user, so i dont have a clue how is that

### VisionCamera Version

react-native-vision-camera: "^2.15.6"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

fukemy, 2023-08-25:
dont do this, if you want to capture photo, just make a customCaptured then convert the frame into Image, save to local then pass it's url back to React Native

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy