margelo/react-native-vision-camera #1712: 🐛 Mass amount of Unresolved reference when update to v3

lninnnn, 2023-08-23:
### What were you trying to do?

Updated from 2.15.4 to 3.0.0-rc.5

### Reproduceable Code

```tsx
"react-native-vision-camera": "3.0.0-rc.5",//tried to use react-native-worklet-core but the package is not linked in cMake and build.gradle yet (still use the old name)
"react-native-vision-camera": "3.0.0-rc.5",
"@shopify/react-native-skia": "^0.1.202",
```

### What happened instead?

Build fail

### Relevant log output

```shell
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081
.../android/src/main/java/com/mrousavy/camera/extensions/MediaFormat.setDynamicRangeProfile.kt: (3, 40): Unresolved reference: DynamicRangeProfiles

../react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/MediaFormat.setDynamicRangeProfile.kt: (34, 54): Unresolved reference: TIRAMISU

react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/MediaFormat.setDynamicRangeProfile.kt: (35, 63): Unresolved reference: FEATURE_HdrEditing
../node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/Size+Extensions.kt: (12, 10): Unresolved reference: minBy

../react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/Size+Extensions.kt: (12, 22): Unresolved reference: it

../react-native-vision-camera/android/src/main/java/com/mrousavy/camera/parsers/VideoStabilizationMode.kt: (5, 48): Unresolved reference: CONTROL_VIDEO_STABILIZATION_MODE_PREVIEW_STABILIZATION

../react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraDeviceDetails.kt: (68, 54): Unresolved reference: TIRAMISU

../react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraDeviceDetails.kt: (69, 48): Unresolved reference: REQUEST_AVAILABLE_CAPABILITIES_DYNAMIC_RANGE_TEN_BIT
../react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CameraDeviceDetails.kt: (70, 49): Not enough information to infer type variable T
```

### Device

Android

### VisionCamera Version

3.0.0-rc05

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

BharatDawai, 2023-08-29:
and progress on this ?

ObidosDev, 2023-09-11:
Same for me
```
node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/Size+Extensions.kt: (12, 10): Unresolved reference: minBy
```

ObidosDev, 2023-09-11:
To fix this need to update `kotlinVersion = '1.7.20'`

lninnnn, 2023-09-12:
> To fix this need to update `kotlinVersion = '1.7.20'`

my build gradle (check above) is already declared kotlinVerson = 1.7.20

lninnnn, 2023-09-12:
@mrousavy Hi! Do you have time to look into this?

mrousavy, 2023-09-13:
Hi! No, why?

keisan1231, 2023-09-14:
`node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/Size+Extensions.kt: (12, 10):`

Unresolved reference: minBy

This error😅

` "react-native-vision-camera": "^3.0.0",`

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

interstates21, 2023-11-15:
It's not fixed