margelo/react-native-vision-camera #1715: 🐛 Unable to resolve "react-native-worklets/src"

sravis02, 2023-08-23:
### What were you trying to do?

I installed react native vision camera 3 (rc5) and created a development build with expo (49.0.8) and expo dev client (2.4.7).

The development build got created successfully but when I am trying to start the app i always get this error.

I installed these two libaries, since I am not sure which is needed. (the error mentions react-native-worklets but on twitter I saw that "-core" should be appended?)
```
    "react-native-worklets": "^0.0.1-alpha",
    "react-native-worklets-core": "^0.2.0"
```

I also added it to my babel.config.js

```
module.exports = function (api) {
  api.cache(true);
  return {
    presets: ['babel-preset-expo'],
    plugins: [
      ['@babel/plugin-proposal-decorators', { legacy: true }],
      ['@babel/plugin-transform-flow-strip-types'],
      ['@babel/plugin-proposal-class-properties', { loose: true }],
      ['react-native-worklets-core/plugin'],
      'react-native-reanimated/plugin',
      require.resolve('expo-router/babel'),
    ],
  };
};
```

### Reproduceable Code

```tsx
/
```

### What happened instead?

[image]

### Relevant log output

```shell
Unable to resolve "react-native-worklets/src" from "node_modules/react-native-vision-camera/src/FrameProcessorPlugins.ts"
```

### Device

iPhone 11

### VisionCamera Version

3.0.0-rc.5

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

marouane-edghoughi, 2023-08-23:
1- Go to the file from where this issue is coming: `node_modules/react-native-vision-camera/src/FrameProcessorPlugins.ts`
2- change the Worklets import from `react-native-worklets/src` to `react-native-worklets-core/src` (make sure you have rn worklets core installed).
Or you can wait for the next update as the developer is currently working on this issue.

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy