margelo/react-native-vision-camera #1716: 🐛 [unknown/unknown]: [unknown/unknown] Error retrieving camcorder profile params

bglgwyng, 2023-08-24:
### What were you trying to do?

Set up RNVC V3 with Expo and tested basic features.

### Reproduceable Code

```tsx
https://github.com/bglgwyng/Expo49RNVC3Sandbox
```

### What happened instead?

```
[unknown/unknown]: [unknown/unknown] Error retrieving camcorder profile params🐛 
```
This error message was emitted. 
I don't believe this is related to the device tested since the bare RN project + RNVC V3 worked very well.

Here is the `nativeError` of it.
```json
{
  "nativeStackAndroid": [
    {
      "lineNumber": -2,
      "file": "CamcorderProfile.java",
      "methodName": "native_get_camcorder_profiles",
      "class": "android.media.CamcorderProfile"
    },
    {
      "lineNumber": 605,
      "file": "CamcorderProfile.java",
      "methodName": "getAll",
      "class": "android.media.CamcorderProfile"
    },
    {
      "lineNumber": 132,
      "file": "CameraDeviceDetails.kt",
      "methodName": "getMaximumVideoSize",
      "class": "com.mrousavy.camera.utils.CameraDeviceDetails"
    },
    {
      "lineNumber": 149,
      "file": "CameraDeviceDetails.kt",
      "methodName": "getVideoSizes",
      "class": "com.mrousavy.camera.utils.CameraDeviceDetails"
    },
    {
      "lineNumber": 166,
      "file": "CameraDeviceDetails.kt",
      "methodName": "getFormats",
      "class": "com.mrousavy.camera.utils.CameraDeviceDetails"
    },
    {
      "lineNumber": 255,
      "file": "CameraDeviceDetails.kt",
      "methodName": "toMap",
      "class": "com.mrousavy.camera.utils.CameraDeviceDetails"
    },
    {
      "lineNumber": 149,
      "file": "CameraViewModule.kt",
      "methodName": "invokeSuspend",
      "class": "com.mrousavy.camera.CameraViewModule$getAvailableCameraDevices$1"
    },
    {
      "lineNumber": 33,
      "file": "ContinuationImpl.kt",
      "methodName": "resumeWith",
      "class": "kotlin.coroutines.jvm.internal.BaseContinuationImpl"
    },
    {
      "lineNumber": 106,
      "file": "DispatchedTask.kt",
      "methodName": "run",
      "class": "kotlinx.coroutines.DispatchedTask"
    },
    {
      "lineNumber": 570,
      "file": "CoroutineScheduler.kt",
      "methodName": "runSafely",
      "class": "kotlinx.coroutines.scheduling.CoroutineScheduler"
    },
    {
      "lineNumber": 750,
      "file": "CoroutineScheduler.kt",
      "methodName": "executeTask",
      "class": "kotlinx.coroutines.scheduling.CoroutineScheduler$Worker"
    },
    {
      "lineNumber": 677,
      "file": "CoroutineScheduler.kt",
      "methodName": "runWorker",
      "class": "kotlinx.coroutines.scheduling.CoroutineScheduler$Worker"
    },
    {
      "lineNumber": 664,
      "file": "CoroutineScheduler.kt",
      "methodName": "run",
      "class": "kotlinx.coroutines.scheduling.CoroutineScheduler$Worker"
    }
  ],
  "userInfo": null,
  "message": "[unknown/unknown] Error retrieving camcorder profile params",
  "code": "unknown/unknown"
}
```

### Relevant log output

```shell
unknown/unknown: [unknown/unknown] Error retrieving camcorder profile params
[unknown/unknown]: [unknown/unknown] Error retrieving camcorder profile params
    at construct (native)
    at apply (native)
    at _construct (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:4486:28)
    at Wrapper (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:4448:25)
    at construct (native)
    at _createSuperInternal (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:131312:322)
    at call (native)
    at CameraError (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:131329:26)
    at construct (native)
    at _createSuperInternal (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:131312:322)
    at apply (native)
    at CameraRuntimeError (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:131387:27)
    at tryParseNativeCameraError (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:131429:38)
    at ?anon_0_ (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:131187:62)
    at throw (native)
    at asyncGeneratorStep (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:6090:26)
    at _throw (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:6112:29)
    at tryCallOne (/root/react-native/packages/react-native/ReactAndroid/hermes-engine/.cxx/Release/4i495j47/arm64-v8a/lib/InternalBytecode/InternalBytecode.js:53:16)
    at anonymous (/root/react-native/packages/react-native/ReactAndroid/hermes-engine/.cxx/Release/4i495j47/arm64-v8a/lib/InternalBytecode/InternalBytecode.js:139:27)
    at apply (native)
    at anonymous (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:39627:26)
    at _callTimer (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:39506:17)
    at _callReactNativeMicrotasksPass (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:39551:17)
    at callReactNativeMicrotasks (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:39757:44)
    at __callReactNativeMicrotasks (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:3696:46)
    at anonymous (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:3470:45)
    at __guard (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:3669:15)
    at flushedQueue (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:3469:21)
    at invokeCallbackAndReturnFlushedQueue (http://172.30.1.20:8081/node_modules/expo/AppEntry.bundle//&platform=android&dev=true&minify=false&app=com.gwyng.Expo49RNVC3Sandbox&modulesOnly=false&runModule=true:3463:33)
```

### Device

Galaxy S10

### VisionCamera Version

3.0.0-rc.5

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bglgwyng, 2023-08-25:
I just tested it with Galaxy S20 and it worked fine. So this issue can be about specific devices.
Also, the stack trace shows that the error came from the OS side.
I've searched the error message  'Error retrieving camcorder profile params' but all I found is abound CameraX, not Camera2.

idrissakhi, 2023-09-12:
@bglgwyng would you share your code with us please ?

bglgwyng, 2023-09-12:
@idrissakhi This issue might be resolved with 3.0.0. But I'm not 100% sure. Which version of RNVC are you using?

idrissakhi, 2023-09-12:
On v3.0.0

bglgwyng, 2023-09-13:
@idrissakhi https://github.com/bglgwyng/Expo49RNVC3Sandbox/commit/d11f80adf9fc1ad4f62aa67264316dd48e1c9967
This commit is where this issue was found.

valn1, 2023-09-28:
logcat:
```
--------- switch to main
09-28 12:01:42.173  2107  2407 E PaniGenerator: network is unknown.
09-28 12:01:42.185  2107  2407 E #IMSCR  : 09-28 12:01:42 0x1101000B:0,FILTERED ALL:11
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.747 14067 14144 E MediaProfiles: The given camcorder profile camera 4 quality 1 is not found
09-28 12:01:42.793 14067 14144 E ReactNativeJS: 'Error: Exception in HostObject::get for prop \'CameraDevices\': java.lang.RuntimeException: Error retrieving camcorder profile params, js engine: hermes', '\n    at Scanner (http://localhost:8081/src/screens/Scanner/index.bundle//&platform=android&app=br.ixcsoft.inmapsales&modulesOnly=true&dev=true&minify=false&runModule=true&shallow=true:14:41)\n    at StaticContainer (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:212523:17)\n    at EnsureSingleNavigator (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:208402:24)\n    at SceneView (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:212428:22)\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:59561:43)\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:59561:43)\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:59561:43)\n    at CardSheet (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:218563:23)\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:59561:43)\n    at anonymous (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:72616:62)\n    at PanGestureHandler (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:123199:38)\n    at PanGestureHandler (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:218644:34)\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:59561:43)\n    at anonymous (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:72616:62)\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:59561:43)\n    at Card (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:218049:36)\n    at CardContainer (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:217670:34)\n    at RNSScreen\n    at anonymous (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:72616:62)\n    at Suspender (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:219264:22)\n    at Suspense\n    at Freeze (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:219283:23)\n    at DelayedFreeze (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:218808:22)\n    at InnerScreen (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inmapsales&modulesOnly=false&runModule=true:218857:36)\n    at Screen (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=br.ixcsoft.inm
09-28 12:01:42.802  2869  2869 E pageboostd: Received HALT command code 2
09-28 12:01:42.817 14067 14144 E ReactNativeJS: TypeError: Cannot read property 'CameraDevices' of undefined

```

valn1, 2023-09-28:
i also got this:
```
09-28 12:16:39.955 17411 17411 E Berry_Drawer: getDrawerListName() : Exception : java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.samsung.accessory.hearablemgr.common.uhm.DeviceRegistryData.deviceId' on a null object reference in method 'java.lang.String com.samsung.accessory.hearablemgr.module.home.drawer.Drawer.getDrawerListName(com.samsung.accessory.hearablemgr.common.uhm.DeviceRegistryData, java.util.List)'

```

valn1, 2023-09-28:
i found out that in the file "node_modules/react-native-vision-camera/src/CameraDevices.ts", NativeModules returns {} for some reason

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy