margelo/react-native-vision-camera #1723: ❓ Version Compatibility Inquiry for Implementing QR Code Reading App with react-native-camera-vision

jgilyam, 2023-08-26:
### Question

Hello! I'm trying to implement an app that requires reading QR codes. I'm encountering several issues with the versions of different dependencies. To use the latest version of react-native-camera-vision (2.15.6), which version of React Native should I use (I couldn't find in the documentation until which version it's supported), and which version of React Reanimated? The documentation mentions 2.2 or higher, but if there's a specific version you would recommend, I'd appreciate your advice. Thank you and best regards.

### VisionCamera Version

2.15.6

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

DeveloperMCD, 2023-08-27:
You've opened a big can of worms. That's what started all troubles for me -- wanting to offer CROSS PLATFORM barcode scanning in my app via Vision Camera. There's a great barcode scanning library that only does Android -- all was well at that point. But when I tried to add iOS support, I had to switch to Vision Camera and that was when all heck broke loose. All kinds of compatibility issues with reanimated, vision-code-scanner, vision camera, react native itself, etc. It's a huge tangled rats nest.

My latest situation: still no X-platform barcode scanning, but at least my app builds successfully again with the very latest RC-6 of 3.0.0 Vision Camera, React Native Worklets Core 0.2.0, React Native 0.72.4, Reanimated 3.4.2, etc.

The main barcode scanner (frame processor) is hopelessly out of date. There is a fork by RKMackinnon, but even that version doesn't build successfully with my current app build (see 2nd paragraph for what versions I'm using). Rumor has it that fork works with "RC3" and/or April 2023 version of Vision Camera. However, I tried RC3 and it didn't work for me!

Here is the latest version of "Vision Camera Code Scanner" -- MacKinnon wasn't familiar with publishing publicly to NPM, so I took care of that chore by forking his repo.

https://www.npmjs.com/package/vision-camera-v3-barcode-scanner

mmatosEKOBIT, 2023-08-29:
@DeveloperMCD Hi, I tried using the "Vision Camera Code Scanner" fork you made, and I followed the instructions to install the package. But when I try to call any method the package is not recognized. 

[image]

jamonholmgren, 2023-08-29:
FYI, I just spoke with @mrousavy and he does intend to work on a plugin for barcode scanning that is compatible with React Native Vision Camera v3. I would expect (although this isn't something Marc has promised) that it will be out in the next few months. So, yes, kind of a gap here, but it will eventually be fixed.

In my personal opinion, it's worth donating some money Marc's way via Github sponsorships (recurring or one-time). I just sent him $300. He's putting a lot of time and expertise into these tools for us to use for free, and this helps him devote time away from his company to build and maintain these tools.

https://github.com/sponsors/mrousavy

rkmackinnon, 2023-08-29:
I will take some time to update my fork of vision-camera-code-scanner at some point. In the meantime, we are currently using the most recent code in vision-camera-code-scanner in a cross platform app with the following in package.json:

```
"react": "18.2.0",
"react-native": "^0.71.7",
"react-native-vision-camera": "github:mrousavy/react-native-vision-camera#820db3c",
"react-native-worklets": "github:chrfalch/react-native-worklets#10abdf1",
"@shopify/react-native-skia": "^0.1.186",
"vision-camera-code-scanner": "file:packages/vision-camera-code-scanner.tgz",
```

With vision-camera-code-scanner.tgz generated by running npm pack. If @DeveloperMCD's repo is an exact copy of mine but just having figured out the public publishing to npm bit, replacing that last line with their package should do the trick for anyone looking to run this.

Development on v3 of react-native-vision-camera is pretty rapid right now and the API for frame processors has been changing between different RCs. As of RC4, a change was made to switch to an object-oriented design for frame processors. The typescript types for the native function return types at that point changed from unknown to ParameterType (defined in react-native-vision-camera), which is a breaking change for vision-camera-code-scanner.

The detailed QR code metadata that Google's MLKit barcode-scanning library returns as complex objects were previously passed directly through to the JS side, which was allowed when the return type was unknown. We are not able to do that anymore without significantly rewriting/mapping that information and creating our own flattened types, which would be incompatible with anyone upgrading from a previous version of vision-camera-code-scanner.

I had mentioned this here: https://github.com/mrousavy/react-native-vision-camera/pull/1466#issuecomment-1663200095, but shortly after that @mrousavy requested we create separate issues for any v3 questions and I haven't done that yet. I will do so. In any case, they have clearly been so incredibly busy working on this pretty incredible upgrade that I think it's understandable that accommodating downstream projects take a bit of a back seat to some of these major architectural changes.

obernardovieira, 2023-08-30:
Or, you could use this https://github.com/mrousavy/react-native-vision-camera/issues/1662#issuecomment-1648804009
Works for both Android and iOS. I use it. No problems so far.

mrousavy, 2023-09-05:
Thanks @jamonholmgren, you're the man. Really appreciate your help here big time ❤️ 

Yea it will be a couple of weeks, not months - will start working on this either next week or the week after 😄

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

hengkx, 2023-09-30:
Do you still need to use this vision-camera-v3-barcode-scanner?

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.