margelo/react-native-vision-camera #1725: Record video in Landscape Mode not working in IOS🐛

HemangiVekaria, 2023-08-28:
### What were you trying to do?

I am using react-native-vision-camera for recording video in Portrait and landscape.
By using react-native-orientation-locker i am locking preview to Landscape or portrait. When click on Landscape button from app camera is opening at that time i lock orientation to landscape so my screen UI rendered in Landscape but preview of camera is showing portrait.

Can anyone please help to record video in landscape. i tried orientations "landscapeLeft",etc but nothing works.

### Reproduceable Code

```tsx
<Camera
                                style={StyleSheet.absoluteFill}
                                ref={cameraRef}
                                device={deviceAndroid}
                                isActive={true}
                                video={true}
                                videoStabilizationMode="standard"
                                enableZoomGesture={true}
                                format={{
                                    videoWidth: getPictureSize().width,
                                    videoHeight: getPictureSize().height,
                                    photoWidth: getPictureSize().width,
                                    photoHeight: getPictureSize().height,
                                    frameRateRanges: [
                                        {
                                            minFrameRate: 24,
                                            maxFrameRate: 30,
                                        },
                                    ],
                                }}
                                audio={true}></Camera>
```

### What happened instead?

app is showing preview in portrait.! [IMG_A4CF2BCA3492-1](https://github.com/mrousavy/react-native-vision-camera/assets/17717369/bdce7ab0-c571-40ee-ad1e-5760f1446700)

### Device

 iPhone X  (IOS 16.6)

### VisionCamera Version

2.15.6

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fedpre, 2023-09-05:
Hi @HemangiVekaria , I was able to change the native code to have a "working" version of the orientation change in iOS for the new  3.0 version. 

Basically, I added a listener to the orientation change and it changes whenever the device changes orientation. It's probably not perfect nor ideal, but it works!

 I can share the code with you (the patch) if you that helps!

HemangiVekaria, 2023-09-05:
> Hi @HemangiVekaria , I was able to change the native code to have a "working" version of the orientation change in iOS for the new 3.0 version.
>
> Basically, I added a listener to the orientation [...]

Hi @fedpre Thanks, Please share patch that will be very helpful for me.

fedpre, 2023-09-05:
> Hi @HemangiVekaria , I was able to change the native code to have a "working" version of the orientation change in iOS for the new 3.0 version.
> Basically, I added a listener to the orientation [...]
[react-native-vision-camera+3.0.0.patch](https://github.com/mrousavy/react-native-vision-camera/files/12518684/react-native-vision-camera%2B3.0.0.patch)

> > I can share the code with you (the patch) if you that helps!
> 
> Hi @fedpre Thanks, Please share patch that will be very helpful for me.

Not sure if this is the best way, but attached is the patch. Let me know if it works!

PrgrmmrOleg, 2023-09-13:
> > Hi @fedpre Thanks, Please share patch that will be very helpful for me.
> 
> Not sure if this is the best way, but attached is the patch. Let me know if it works! [react-native-vision-camera+3.0.0.patch](https://github.com/mrousavy/react-native-vision-camera/files/12518684/react-native-vision-camera%2B3.0.0.patch)

Works fine for me. Thank you. You saved my deadline

fedpre, 2023-09-13:
> Works fine for me. Thank you. You saved my deadline

Glad that it worked and helped you!

HemangiVekaria, 2023-09-13:
> Not sure if this is the best way, but attached is the patch. Let me know if it works! [...]
 
After adding this patch in andorid camera is not showing. it shows black screen.   ` <Camera
                    style={StyleSheet.absoluteFill}
                    ref={cameraRef}
                    device={deviceAndroid}
                    isActive={true}
                    video={true}
                    enableZoomGesture={true}
                    format={{
                        videoWidth: getPictureSize().width,
                        videoHeight: getPictureSize().height,
                        photoWidth: getPictureSize().width,
                        photoHeight: getPictureSize().height,
                        frameRateRanges: [
                            {
                                minFrameRate: 24,
                                maxFrameRate: 30,
                            },
                        ],
                    }}
                    audio={true}></Camera>`

mrousavy, 2023-09-30:
Closing as this is a stale issue - this might have been fixed with the full rewrite in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

JFYI; Orientation is a bit more tricky - if I get enough sponsors I can work on this. Camera sensors are always 90deg

closed 2023-09-30 by mrousavy

mrousavy, 2023-09-30:
Hey - I'm tracking Orientation in this feature request/issue now: https://github.com/mrousavy/react-native-vision-camera/issues/1891 ✨