margelo/react-native-vision-camera #1726: [V3] Flash is not working on Android

jthure, 2023-08-29:
### What were you trying to do?

I'm trying to capture a photo with `flash: 'on'`.

### What happened instead?

When trying it on Android, the flash simply doesn't fire. It works fine on iOS

### Relevant log output

```shell
I added a log statement in `CameraSession.kt` to log `result[CaptureResult.FLASH_STATE]`. If I understand the docs correctly it should report [FLASH_STATE_FIRED](https://developer.android.com/reference/android/hardware/camera2/CameraMetadata#FLASH_STATE_FIRED) if the flash was fired, but it does report [FLASH_STATE_READY](https://developer.android.com/reference/android/hardware/camera2/CameraMetadata#FLASH_STATE_READY)
```

### Device

Samsung galaxy S10

### VisionCamera Version

3.0.0-rc.8

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-08-30:
yup, I didn't implement Flash on Android yet. It's a bit more tricky than I thought.

pohsiu, 2023-09-27:
Same condition here

mrousavy, 2023-09-30:
Can you please create a feature request for this? Thanks!

closed 2023-09-30 by mrousavy

mrousavy, 2023-09-30:
Hey - I'm tracking Flash in this feature request/issue now: #1890  ✨ 

Make sure to upvote or sponsor to support this feature, and leave a comment if you have any additional thoughts/ideas.